dji-sdk/Onboard-SDK-ROS #304: What is "Controller Version "?

erenaud3, 2019-10-17:
Hi,

I have a simple question for once: what is "controller version" ? Is that DJI GO app version ?

Thanks

dji-dev, 2019-10-18:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28960'>Zendesk ticket #28960</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Where did you see this word? It may be expressed differently in different document locations.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-18:
I saw it in the ReadMe.md from this current repository. Column at the very middle: 

| Aircraft/FC           | Firmware Package Version | Flight Controller Version | OSDK Branch            | Notes                                                                 |
|-----------------------|--------------------------|---------------------------|------------------------|-----------------------------------------------------------------------|
| **M210/M210 RTK V2**  | **1.0.0450**             | **3.4.3.31**              | **OSDK 3.8.1**         |                                                                       |
|                       |                          |                           |                        |                                                                       |
| **M210/M210 RTK**     | **1.2.0440**             | **3.3.10.12**             | **OSDK 3.8.1**         |                                                                       |
|                       |                          |                           |                        |                                                                       |
| **M600/M600 Pro**     | **1.0.1.66**             | **3.2.41.13**             | **OSDK 3.8.1**         |                                                                       |
|                       |                          |                           |                        |

dji-dev, 2019-10-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28960'>Zendesk ticket #28960</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The full statement should be “Flight Controller Version” .
You can see this flight control version information in the initial log when running OSDK.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-10-21:
Okay, thank you.

closed 2019-10-21 by erenaud3

dji-dev, 2019-10-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28960'>Zendesk ticket #28960</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It's our pleasure to answer your questions, good luck with your project.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team