dji-sdk/Onboard-SDK-ROS #305: Ground station protocol for M600pro

erenaud3, 2019-10-18:
Hi,

Here are my specs:
**Drone: M600pro**
**Firmware: 1.0.1.67**
**System and ROS Version: Ubuntu 16.04 and ROS Kinetic**
**OSDK version: 3.8.1**

I just spotted [ground station protocol for M100](https://developer.dji.com/onboard-sdk/documentation/protocol-doc/ground-station-protocol.html#data-structure). Can I use the same protocol for M600pro ? I would be especially interested in actions that can be used to fill `commandList` from WayPointSettings (which is call WayPointData in this page...).

dji-dev, 2019-10-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28988'>Zendesk ticket #28988</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

1. Can I use the same protocol for M600pro?
==> Sorry, this protocol is not supported DJI M600 Pro.
 [image]​

2. I would be especially interested in actions that can be used to fill `commandList` from WayPointSettings (which is call WayPointData in this page...).
==> About the CommandList represents the list of waypoint actions. CommandParameter represents the list of parameters for a waypoint. For example, the code that needs to hover for 5 seconds after reaching the waypoint:
 [image]​
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/kAXkt1yyntD5wn92qU2poe6mZ' target='zd_attach'>inline632516315.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/oRJlpLpVNp4DPL1PjFzyG6yYJ' target='zd_attach'>inline-1338828910.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/EZuz2726DRn9gMnbobnIqquiq' target='zd_attach'>inline501142186.png</a>

erenaud3, 2019-10-21:
Thank you for your reply.

So, is the code you put in "2." is supposed to work for M600pro eventually ?
Because I can't manage to get it work.

dji-dev, 2019-10-22:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/28988'>Zendesk ticket #28988</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

So, is the code you put in "2." is supposed to work for M600pro eventually ? Because I can't manage to get it work.
==> Sorry that I do not particularly understand the problem you have described. suggest you click the hyperlink on the right to fill in and submit more complete information about the problem, https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-12-10:
Hello again,

If any one is wondering: yes it is possible to set waypoint actions with M600Pro.

dji-dev's answers mislead me by thinking that I only have to modify the function "setWaypointDefaults(...)" from demo_mission.cpp to get a working example. But you also have to modify the function uploadWaypoints(...) from the same file. Otherwise, the actions you set in "setWaypointDefaults" are not copied.

Hope that helps !

closed 2019-12-16 by erenaud3