dji-sdk/Onboard-SDK-ROS #306: Can not call service of setup_camera_stream using M210,Manifold 2-G and xt2.

potBagMeat, 2019-10-23:
Hi,guys!
Today I encountered a problem. I turn on the RC and run
`roslaunch dji_sdk sdk.launch`
`rosservice call /dji_sdk/setup_camera_stream  "{cameraType: 1, start: 1}"`
[image]
[image]
but when I want to subscribe to the fpv camera, everything goes right.
[image]
While I run the linux sample which is called 
> camera-stream-callback-sample

everything was right and displayed the main camera image.

I wonder why this happened and how to fix this problem.
Thanks for help!

dji-dev, 2019-10-23:
Public comment from DJI SDK in <a href='https://djisdksupport.zendesk.com/agent/tickets/29114'>Zendesk ticket #29114</a>:<br><br>Hi,guys!
Today I encountered a problem. I turn on the RC and run
`roslaunch dji_sdk sdk.launch`
`rosservice call /dji_sdk/setup_camera_stream  "{cameraType: 1, start: 1}"`
[image]
[image]
but when I want to subscribe to the fpv camera, everything goes right.
[image]
While I run the linux sample which is called 
> camera-stream-callback-sample

everything was right and displayed the main camera image.

I wonder why this happened and how to fix this problem.
Thanks for help!

pavankumarbn, 2019-10-23:
@potBagMeat  did you try with root privileges?

potBagMeat, 2019-10-23:
@dji-dev sorry, I can't load the page, can you post it here, thanks.

potBagMeat, 2019-10-23:
@pavankumarbn I tried it and it was same result..

dji-dev, 2019-10-24:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29114'>Zendesk ticket #29114</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

In order to better understand and help you solve the problem, need to know more information as follows:
1. M210 model and firmware version
2. The firmware version of XT2
3. OSK - ROS version

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

anish-bhattacharya, 2019-10-25:
I am having this same exact problem. `camera-stream-callback-sample` works for streaming fpv and main camera, but launching `roslaunch dji_sdk sdk.launch` and using the rosservice call works for fpv and fails for main camera, with the exit -11 error. Any help would be greatly appreciated!

Information:
M210 V2
Manifold 2-C
OSDK-ROS 3.8.1
Drone firmware 1.0.0590
X4S gimbal connected to port I of dual gimbal mount

What I have tried:
- Refreshing firmware
- Building with and without advanced sensing flags
- With and without timesync and broadcast functionality set in the sdk.launch file. If both are changed to `true` then I get the error `unable to read from MAIN_CAMERA lost, retry connecting...`
- Both baudrate 921600 and 1000000, changing in DJI Assistant 2 and the sdk.launch file (in both cases the sample demo works)

Another issue that may or may not be related, is that the main camera feed on the CrystalSky is not showing up. It is either black or greyed out and not updating the image. It also gives a warning: "Zenmuse x4s remaining sd card capacity is not enough". If the USB-A connector is disconnected then the feed works again, and it also works while running the `camera-stream-callback-sample` with the main camera. I have tried using another X4S gimbal, but I have the same problems. Thanks in advance!

potBagMeat, 2019-10-29:
@dji-dev 
hey, thank you, I already fix this problem.
Sorry for the late reply.

potBagMeat, 2019-10-29:
@anish-bhattacharya 
hey, bro.
Sorry to reply so late.
I already fix this problem.
If you can run `camera-stream-callback-sample` with `bin/camera-stream-callback-sample UserConfig.txt /dev/ttyACM0`, I will have a solution for you.
The reason why the device displayed 'sd card capacity is not enough' is that the computer recognize your drone as an usb device. In order to avoid that, you should make your computer recognize your drone as an camera device. 
In the linux sample, they use these codes to do that.
[image]
While in ROS version, you can simply add `vehicle->advancedSensing->setAcmDevicePath("/dev/ttyACM0")` in the end of the constructor of the dji_sdk_node.cpp. And don't forget to include the necessary header  file.

dji-dev, 2019-10-29:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29114'>Zendesk ticket #29114</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

@potBagMeat, Congratulations on solving the problem, and thank you for your solution.

@anish-bhattacharya, If it is normal for you to use the Linux sample, it should be the lack of specified USB devices on ROS. It is suggested that you can try to use the solution of @potBagMeat. In addition, we will also analyze and verify the problem and solution.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

anish-bhattacharya, 2019-10-30:
@dji-dev @potBagMeat Thanks a lot for the solution, adding the ttyACM0 line to our constructor solved our gimbal camera streaming problem!

closed 2019-10-31 by potBagMeat

diogon01, 2020-09-23:
> @dji-dev @potBagMeat Thanks a lot for the solution, adding the ttyACM0 line to our constructor solved our gimbal camera streaming problem!
How to add ttyACM0 line in constructor ? This possible you shared your code?  For the reason that I don't have much knowledge in C ++ development.

Thank's so mutch.