dji-sdk/Onboard-SDK-ROS #307: Can not subscribe to RTK data M600

maxhofidrone, 2019-10-29:
Hi,
I tried to subscribe to the RTK data for some time now. Unfortunately I was not able to subscribe successfully yet.
Following line in dji_sdk_node.cpp (511):
`if (vehicle->subscribe->initPackageFromTopicList(PACKAGE_ID_5HZ, nTopicRTKSupport,                                               topicRTKSupport, 1, 10))`
always return false. Therefore the RTK data will not be available later. 
I saw that the frequency may should be changed to 5Hz in  `initPackageFromTopicList` (https://github.com/dji-sdk/Onboard-SDK-ROS/pull/231). But this also did not help. I set the RTK data transmission rate in DJI Assistant to 10 Hz. Furthermore I'm using following workaround (ref. to output of `sdk.launch`):
 

- Ubuntu 18.04, ROS Melodic: ` * /rosdistro: melodic
                                                   * /rosversion: 1.14.3`

- `STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.2.41.14`

- DJI Onboard SDK ROS 3.8.1

- RTK is available (green lights at the modules + visible at DJI GO)

Was anybody able to subscribe to RTK successfully within ROS? If yes, how?

Thanks a lot for any help in advance!
Max

dji-dev, 2019-10-30:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

OSDK-ROS master branch has fixed the wrong frequency in the RTK subscription package, it is recommended that you download the master branch for compilation and testing. https://github.com/dji-sdk/Onboard-SDK-ROS/tree/master

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-10-30:
Dear @dji-dev ,
I just tried the OSDK-ROS master branch. Unfortunately the subscriptionto RTK also fails. Please have a look on the output of `roslaunch dji_sdk sdk.launch` below. The app and RTK modules indicated RTK availability. 

 Does the RTK data transmission setting in DJI Assistant 2 remains on 10 HZ?

Thank you very much for any help!

Best regards,

Max 

STATUS/1 @ activate, L1411: Activation successful

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ initGimbal, L926: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ initGimbal, L970: Gimbal not mounted!
[ INFO] [1572430565.403231257]: drone activated
[ INFO] [1572430565.461357670]: Use data subscription to get telemetry data!
[ INFO] [1572430565.461432313]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 2 result: 0.
STATUS/1 @ startPackage, L355: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L353: Start package 1 result: 0.
STATUS/1 @ startPackage, L355: Package 1 info: freq=50, nTopics=10.

STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=5, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=5, nTopics=11.

STATUS/1 @ startPackage, L353: Start package 3 result: 0.
STATUS/1 @ startPackage, L355: Package 3 info: freq=400, nTopics=1.

ERRORLOG/1 @ getValue, L315: Topic 0x26 value memory not initialized, return default[FATAL] [1572430565.808369634]: ASSERTION FAILED
	file = /opt/ros/melodic/include/ros/publisher.h
	line = 102
	cond = false
	message = 
[FATAL] [1572430565.808558751]: Call to publish() on an invalid Publisher
[FATAL] [1572430565.808676987]: 

[dji_sdk-2] process has died [pid 6257, exit code -5, cmd /home/kanaria/ROS_DJI_unchanged/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/home/kanaria/.ros/log/43fee048-fafe-11e9-9c1c-33735725c17b/dji_sdk-2.log].
log file: /home/kanaria/.ros/log/43fee048-fafe-11e9-9c1c-33735725c17b/dji_sdk-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

dji-dev, 2019-11-04:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Max,

Thank you for contacting DJI.

DJI OSDK-ROS in the normal case of RTK, the master node will subscribe RTK data (5HZ), can be referred to https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L512, So you don't need to set the RTK frequency in DJI Assistant2.
It is recommended that you can check and confirm that the frequency defined in your OSDK-ROS program is also 5HZ.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-11-04:
Dear @dji-dev,
I can confirm, that the frequency in my OSDK-ROS is defined as 5Hz: [L512]( https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L512).
Unfortunately the master node is not able subscribe RTK due to a memory issue which can be seen in the output of `sdk.launch` above!
 I did not change any code of the OSDK-Master after cloning it to my PC. Can you subscribe to RTK with this branch? It would be exremly important for me to use RTK with the OSDK-ROS.

Thank you very much and best regards,

Max

dji-dev, 2019-11-05:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Max,

Thank you for contacting DJI.

What RTK equipment are you using on the M600? DJI D-RTK?
Also, according to the log analysis you provided, OSDK-ROS seems unable to get into the “if” code location.
It is suggested that you can add debug information to the code location for judgment.
After the OSDK-ROS node is started, can you see the relevant RTK TOPIC?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-11-06:
Dear  @dji-dev ,
I am using the DJI-D-RTK ([here](https://www.dji.com/de/d-rtk/info)).

The `if` code in [L516](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L516) is true. Also the code is running the `else` condition in [L525](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L525). 
I included some specific outputs of the related variables in the dji_sdk_node.cpp ([dji_sdk_node.cpp.txt](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3815186/dji_sdk_node.cpp.txt)). After starting `sdk.launch` (including the outputs mentioned above: [error_sdk.launch.txt](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3815213/error_sdk.launch.txt)) I am able to see the RTK Topics with `rostopic list`. But it seems as the actual topics are not published. You can find the output here: [error_rostopic.txt](https://github.com/dji-sdk/Onboard-SDK-ROS/files/3815209/error_rostopic.txt).  It is  also suspicious, as I can not access the `/dji_sdk/gps_position` in this case.

Do you need any more specific debug information? Are you able to subscribe the RTK data with my provided `dji_sdk_node.cpp`?

Thank you very much in advance,

Best regards,

Max

dji-dev, 2019-11-07:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Max,

Thank you for contacting DJI.

Notice the "Topic 0x26 value memory not initialized" error message in OSDK-ROS node, is the probability of this error 100%?
It is recommended to restart the aircraft and OSDK equipment, whether this error can be eliminated? need to ensure that the OSDK-ROS node does not report relevant error information when it starts.
In addition, the emergence of RTK related topics indicates that OSDK-ROS can be properly identified to RTK devices, it is recommended that using the baud rate of 921600 or 1M in the ROS environment and retry subscribing to RTK topic information.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

maxhofidrone, 2019-11-07:
Dear @dji-dev ,
this error appears all the time for 100%. The only way to avoid it, is to set the frequency back to 10 Hz in [L516/L517](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L516) which will then result into not subscribing to RTK.
The baud rate is set to 921600 and any restarts of the aircraft and OSDK equipment has no positive effect to avoid this error. For me it seems like a bug inside the OSDK to publish the RTK topics correctly into the ROS framework. Could you successfully subscribe to RTK with ROS for a comparable setup (M600 ROS Melodic)? 

It is mandatory for my project to use RTK and therefore a fix of this issue soon would be appreciated a lot!  

Thank you very much and best regards,

Max

dji-dev, 2019-11-08:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Max,

Thank you for contacting DJI.

After the actual test and verification, we cannot repeat your problem.
Suggest that you can use the test demo provided in the attachment for verification.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/ZMwuOakcqm8GWSsarqprZwKmu' target='zd_attach'>catkin_600.tar.gz</a>

maxhofidrone, 2019-11-11:
Dear @dji-dev,
thank you very much for providing this test demo. It worked without any problem to subscribe the RTK topics!
I also found the error in `dji_sdk_node.cpp`. The subscription/advertising frequency of the RTK related advertiser in lines [L544](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node.cpp#L544)-[L557](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node.cpp#L557) needed to be fixed to 5Hz! This is missing in version 3.8 and the [master](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/dji_sdk/src/modules/dji_sdk_node.cpp#L550) branch.

Thanks again for the great help with this issue!

Best regards,

Max

dji-dev, 2019-11-12:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29314'>Zendesk ticket #29314</a>:<br><br>Dear Max,

Thank you for contacting DJI.

Congratulations on solving the problem and thank you for your feedback.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2019-11-12 by maxhofidrone

tkazik, 2020-09-26:
After spending a few hours on that, some comments:
- This [commit](https://github.com/dji-sdk/Onboard-SDK-ROS/commit/594df25d7ef43c9b11994899d3aabaf88f7bde19), which has recently been merged with this [PR](https://github.com/dji-sdk/Onboard-SDK-ROS/commit/705d9c364858497e4ddecae5b7d9d168fa8c1b5b), actually reintroduces the above error: `L315: Topic 0x26 value memory not initialized`
- As @maxhofidrone points out, lines [L544](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/594df25d7ef43c9b11994899d3aabaf88f7bde19/dji_sdk/src/modules/dji_sdk_node.cpp#L544) to [L557](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/594df25d7ef43c9b11994899d3aabaf88f7bde19/dji_sdk/src/modules/dji_sdk_node.cpp#L557) should probably also be set to 5Hz
- On line [L565](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/594df25d7ef43c9b11994899d3aabaf88f7bde19/dji_sdk/src/modules/dji_sdk_node.cpp#L565), the frequency of the topic TOPIC_RTK_CONNECT_STATUS should probably be set to 50Hz (see [here](https://developer.dji.com/onboard-api-reference/group__telem.html#ga563d2333dc43253129a3915970f99cfaa622484559b4778ae49d9557182cf15e0))

The tar-file from this comment [above](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/307#issuecomment-551627582) fixes the error (`L315: Topic 0x26 value memory not initialized`), but also uncomments lines [L276-L277](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/705d9c364858497e4ddecae5b7d9d168fa8c1b5b/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L276) and lines [L306-L308](https://github.com/dji-sdk/Onboard-SDK-ROS/blob/705d9c364858497e4ddecae5b7d9d168fa8c1b5b/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L306), which has the following consequence:

- The topic TOPIC_RTK_CONNECT_STATUS is not published

The current status of the sdk version 3.8 is thus not running smoothly out of the box and DJI should fix it further.
My two cents...hoping that this will help someone in the future.

snakehaihai, 2022-08-11:
I can confirm that the dji RTK is only output at 5hz 

[image]

mwunguzi, 2023-09-14:
I have had similar issues when launching the sdk.launch file, to fix the error go to ONBOARD-SDK-ROS-3.8/dji_sdk/src/modules/dji_sdk_node.cpp and comment out all code related to TOPIC_RTK_POSITION. Those are from L510-L530 and L540-L572.  I am using OSDK-ROS-3.8 with M600 Pro. Below is a screenshot of the error I was getting when launching the sdk.launch file. Hope this helps someone! 
[image]