dji-sdk/Onboard-SDK-ROS #308: roslaunch dji_sdk failure

Ebtiisam, 2019-11-04:
i tried to launch the sdk package by typing

`< roslaunch dji_sdk sdk.launch >`

however every time i get this error:

... logging to /home/dji/.ros/log/b31d5e94-fec0-11e9-9835-74da38ebc18f/roslaunch-manifold2-18825.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:41911/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1079440
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: 39d353a8e9ad01b6c...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [18835]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b31d5e94-fec0-11e9-9835-74da38ebc18f
process[rosout-1]: started with pid [18848]
started core service [/rosout]
process[dji_sdk-2]: started with pid [18856]

STATUS/1 @ init, L56: Attempting to open device /dev/ttyUSB0 with baudrate 921600...

STATUS/1 @ init, L66: ...Serial started successfully.

ERRORLOG/1 @ getDroneVersion, L1503: Drone version not obtained! Please do not proceed.
Possible reasons:
Serial port connection:
* SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
* Baudrate is not correct, please double-check from DJI Assistant2 -> SDK -> baudrate.
* TX and RX pins are inverted.
* Serial port is occupied by another program.
* Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username). Then logout and login again

ERRORLOG/1 @ activate, L1387: Unable to initialize some vehicle components![ERROR] [1572843926.785053218]: drone activation error
[ERROR] [1572843926.785190956]: Vehicle initialization failed
^C[dji_sdk-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

here is the step that i followed to activate the drone: 
`<
catkin_make
source devel/setup.bash
rosed dji_sdk sdk.launch 
>`
where i edit the baud rate/app id/ enc key/ and serial name
app_id: 1079440
baud_rate: 921600
serial_name: /dev/ttyUSB0
enc_key: 39d353a8e9ad01b6c659111904bae57ebb138785aa5159699383f28da3f56563

and i continued with the following commands: 
`<
roslaunch dji_sdk sdk.launch 
>`
its where i got the error

i'm using USB-TTL cable to connect manifold 2-c with M210 RTK V2, also i got the permission first before typing any code by typing :

`<usermod -a -G dialout $USER >`
and then log out and log in 
then i typed :
  `< sudo chmod 666 /dev/ttyUSB0>`
to read and give permission to serial port.

TX (green wire) and Rx (white wire) and ground (black wire) there were put respectively.

so now what should i do?

Regards,

pavankumarbn, 2019-11-04:
@Ebtiisam I also faced a similar problem, in my case TX and RX  inverted, RC was not set to 'F' Mode.  I have detailed in this link:  https://pavankumarbn.files.wordpress.com/2018/05/how-to-activate-matrice-100-drone-with-intel-nuc-as-onboard-computer.pdf
-BW
Pavan

Ebtiisam, 2019-11-04:
i tried to invert RX and TX but i got the same error

pavankumarbn, 2019-11-04:
@Ebtiisam How about baud rate? did you set the same baud rate for a serial port?

Ebtiisam, 2019-11-04:
baud_rate: 921600 as it was mentioned in https://github.com/dji-sdk/Onboard-SDK-ROS

Ebtiisam, 2019-11-04:
@pavankumarbn baud_rate: 921600 as it was mentioned in https://github.com/dji-sdk/Onboard-SDK-ROS

pavankumarbn, 2019-11-04:
@Ebtiisam Yes, Did you set the same baud rate for USB-TTL cable?

Ebtiisam, 2019-11-04:
@pavankumarbn how can i set baud rate for  USB-TTL cable?

pavankumarbn, 2019-11-04:
> @pavankumarbn I cannot open assistant 2 i tried using two laptops but it does not open when i connect it to the drone. also what do you mean by RC and mode (f)??

Please go through this link for more details about different modes: https://developer.dji.com/mobile-sdk/documentation/introduction/component-guide-remotecontroller.html

pavankumarbn, 2019-11-04:
> @pavankumarbn how can i set baud rate for USB-TTL cable?

I detailed  in this document ( https://pavankumarbn.files.wordpress.com/2018/05/how-to-activate-matrice-100-drone-with-intel-nuc-as-onboard-computer.pdf), 

To make you easy: https://askubuntu.com/questions/592386/ubuntu-putty-and-serial-port?utm_medium=organic&utm_source=google_rich_qa&utm_campaign=google_rich_qa 

Follow this link step by step

Ebtiisam, 2019-11-04:
i'm using M210 drone and there is no mode F in the remote control

Ebtiisam, 2019-11-04:
@pavankumarbn should i use sudo putty /dev/ttyUSB0 -serial -sercfg 9600,8,n,1,N or minimum for the serial port?

pavankumarbn, 2019-11-04:
@Ebtiisam I suggest to use minicom.

pavankumarbn, 2019-11-04:
> i'm using M210 drone and there is no mode F in the remote control

Ok.

Ebtiisam, 2019-11-04:
@pavankumarbn So i did the step by step as you advised and when i tried to launch the package again i got the same error

pavankumarbn, 2019-11-04:
@Ebtiisam, Please share the serial port configuration setup details here?

Ebtiisam, 2019-11-04:
@pavankumarbn  use minicom. To install minicom use command
sudo apt-get install minicom
Start minicom setup with command
sudo minicom -s
You will get setup menu
go to Serial port setup
Change Serial Device
Change Bps/Par/Bits
After changes hit Enter
Save setings
Go to Exit to exit from setup in minicom

AS for the connection as i mentioned above i used USB-TTL cable

pavankumarbn, 2019-11-04:
[image]

Ebtiisam, 2019-11-04:
@pavankumarbn yes exactly

pavankumarbn, 2019-11-04:
@Ebtiisam If possible please share a recorded video of your entire process step by step to this email id pavanbn8@gmail.com, I will get back to you.

Ebtiisam, 2019-11-04:
@pavankumarbn i will

dji-dev, 2019-11-05:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29511'>Zendesk ticket #29511</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you follow the steps below to troubleshoot the problem:
1. It is recommended that you can first confirm whether the USB-TTL device can work normally at 921600 baud rate, such as using a serial port assistant (software tools) for testing.  And suggest that use the USB-TTL IC of FT232.
2. In addition, please refer to the user manual of M210 V2 and connect the serial line according to the definition of the UART interface.
3. SDK is not enabled, please check DJI Assistant2 -> SDK -> [v] Enable API Control.
4. Baudrate is correct? please double-check from DJI Assistant2 -> SDK -> baudrate. It should be consistent with the OSDK-ROS configuration file.
5. TX and RX pins are inverted.
6. The serial port is occupied by another program.
7. Permission required. Please do 'sudo usermod -a -G dialout $USER' (you do not need to replace $USER with your username).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

Ebtiisam, 2019-11-05:
@dji-dev  

(1)- how can i confirm, do you mean like minicom? 
(3-4)- i cannot open dji assistant 2 in any laptop, it does not connect. 
(5)- TX and RX are not inverted because they are placed such as the figure in this https://developer.dji.com/onboard-sdk/documentation/M210-Docs/aircraft-checklist.html
(6)- How is that ? is there any way to know?
(7)- i have already done that.

Ebtiisam, 2019-11-05:
and when i tried baud rate 230400 same like in assistant 2 i got this error 

STATUS/1 @ init, L56: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 1DADG7200100F1

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = PM420

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.4.3.31

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x304031F

STATUS/1 @ getErrorCodeMessage, L752: activate
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1424: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1431: Solutions for NEW_DEVICE_ERROR:
* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
* If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
For different aircraft, the App and the version of DJI Assistant 2 might be different
For A3, N3, M600/Pro and M100, please use DJI GO App
For M210 V1, please use DJI GO 4 App or DJI Pilot App
For M210 V2, please use DJI Pilot App
For DJI Assistant 2, it's available on the 'Download' tab of the product page
* If this device is previously activated with another app_id and app_key, you will need to re-activate it again.
* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.
[ERROR] [1572940707.785004134]: drone activation error
[ERROR] [1572940707.785109778]: Vehicle initialization failed
^C[dji_sdk-2] killing on exit

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: reset
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Ebtiisam, 2019-11-05:
i'm using MacBook and assistant2  opened once when i used an adaptor a month ago and i got this baud rate 230400 when i tried it i got this error ;

... logging to /home/dji/.ros/log/0b48f920-ffa2-11e9-8ad0-74da38ebc18f/roslaunch-manifold2-9334.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:36929/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1079440
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: 39d353a8e9ad01b6c...
 * /dji_sdk/serial_name: /dev/ttyUSB0
 * /dji_sdk/use_broadcast: False
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [9344]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0b48f920-ffa2-11e9-8ad0-74da38ebc18f
process[rosout-1]: started with pid [9357]
started core service [/rosout]
process[dji_sdk-2]: started with pid [9367]

STATUS/1 @ init, L56: Attempting to open device /dev/ttyUSB0 with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 1DADG7200100F1

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = PM420

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.4.3.31

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x304031F

STATUS/1 @ getErrorCodeMessage, L752: activate
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1424: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1431: Solutions for NEW_DEVICE_ERROR:
* Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
* If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
For different aircraft, the App and the version of DJI Assistant 2 might be different
For A3, N3, M600/Pro and M100, please use DJI GO App
For M210 V1, please use DJI GO 4 App or DJI Pilot App
For M210 V2, please use DJI Pilot App
For DJI Assistant 2, it's available on the 'Download' tab of the product page
* If this device is previously activated with another app_id and app_key, you will need to re-activate it again.
* A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.
[ERROR] [1572940707.785004134]: drone activation error
[ERROR] [1572940707.785109778]: Vehicle initialization failed
^C[dji_sdk-2] killing on exit

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: reset
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

dji-dev, 2019-11-06:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29511'>Zendesk ticket #29511</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It can be executed under 230400 with the prompt "NEW_DEVICE_ERROR", which is a normal phenomenon, and you can activate the operation according to the prompt of the log.
However, it is recommended to use the baud rate of 921600 in the ROS environment, so you need to set the baud rate of 921600 in DJI Assistant2 for Matrice.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

Ebtiisam, 2019-11-17:
Since DJI assistant is not connecting, i think i will try to execute 230400, how can i activate it since i get this error 

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x304031F

STATUS/1 @ getErrorCodeMessage, L752: activate
STATUS/1 @ getCMDSetActivationMSG, L801: NEW_DEVICE_ERROR

ERRORLOG/1 @ activate, L1424: Failed to activate please retry SET 0x0 ID 0x1 code 0x3

ERRORLOG/1 @ activate, L1431: Solutions for NEW_DEVICE_ERROR:

Double-check your app_id and app_key in UserConfig.txt. Does it match with your DJI developer account?
If this is a new device, you need to activate it through the App or DJI Assistant 2 with Internet
For different aircraft, the App and the version of DJI Assistant 2 might be different
For A3, N3, M600/Pro and M100, please use DJI GO App
For M210 V1, please use DJI GO 4 App or DJI Pilot App
For M210 V2, please use DJI Pilot App
For DJI Assistant 2, it's available on the 'Download' tab of the product page
If this device is previously activated with another app_id and app_key, you will need to re-activate it again.
A new device needs to be activated twice to fix the NEW_DEVICE_ERROR, so please try it twice.
[ERROR] [1572940707.785004134]: drone activation error
[ERROR] [1572940707.785109778]: Vehicle initialization failed
^C[dji_sdk-2] killing on exit
STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: removePackage
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH

STATUS/1 @ getErrorCodeMessage, L752: reset
STATUS/1 @ getCMDSetSubscribeMSG, L835: SUBSCRIBER_ILLEGAL_DATA_LENGTH
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

Ebtiisam, 2019-11-26:
I opened Assistant 2 using adaptor, i changed the baud rate and enabled the API, but i get a new device error

gouyiyong, 2020-03-17:
> I opened Assistant 2 using adaptor, i changed the baud rate and enabled the API, but i get a new device error

Hi, i meet the same issue NEW_DEVICE_ERROR, do you find any solution for this?

Ebtiisam, 2020-05-13:
No still.

HuangDihe, 2021-01-15:
I got the  same  problem  with  A3.    But  I   got  a  lot   of  data  in the serial usb.

dji-kyle, 2021-07-08:
NEW_DEVICE_ERROR, using DJI Assistant2 to activate the APPID，running OSDK application twice with the DJI Assitant2 connected.

closed 2021-07-08 by dji-kyle