dji-sdk/Onboard-SDK-ROS #309: Does this SDK support control of velocity of three axises and yaw angle simultaneously?

MicroYY, 2019-11-11:
I have a requirement that the velocity of three axises and the yaw angle(not yaw rate) should be taken as input. 
Velocity control is supported in `controlVelYawRate`, but yaw angle is not.
`controlPosYaw` supports yaw angle control, but i need to specify a position, which conflicts with `controlVelYawRate` that velocity control does not work.
Could you provide a solution?

maxhofidrone, 2019-11-11:
Hi!
as far as I understand your needs, I think it should possible with the 
`/dji_sdk/flight_control_setpoint_generic` flight control topic. You can decide with the `FlightControlFlag` how your input should be handled (velocity + angle).

Best regards,
Max

MicroYY, 2019-11-11:
> Hi!
> as far as I understand your needs, I think it should possible with the
> `/dji_sdk/flight_control_setpoint_generic` flight control topic. You can decide with the `FlightControlFlag` how your input should be handled (velocity + angle).
> 
> Best regards,
> Max

Oh.  I have tried this, and it really works!
Thanks!!!    ^_^

closed 2019-11-11 by MicroYY