dji-sdk/Onboard-SDK-ROS #310: What is the point of "/dji/time_trigger" topic ?

erenaud3, 2019-11-13:
Hi !

I thought "dji/time_trigger" was supposed to hold FC's time at which the hardware pulse was sent (during hardware synchronization), from what I read on the [documentation](https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-hardware-sync.html) and from what I understood in this [issue](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/298). So I don't understand at all the following code from `dji_sdk_node_publisher.cpp`:

https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L630-L701

In the last condition, we can see that if the hardSync packet we just received corresponds to a hardware pulse (`if (hardSync_FC.ts.flag == 1)`), then a message is published on `dji/trigger_time` (thanks to `p->trigger_publisher`) with `now_time` as both timestamp and time_ref.

So basically, what we are doing here is publishing the timestamp at which **ROS** received the hard sync packet. In other words, we have the ROS time at which hard sync packet was received (reminder: the topic is supposed to hold the time at which **FC** sent the packet).

Not only this information is useless for hardware synchronization but also it is a non-sense. In order to work, it would imply that hard sync packet is instantly send by FC and received by OC. So why would we need hardware synchronization in the first place ?

I don't have any workaround to propose. My first guess would have been to do something like:
```cpp
  if (hardSync_FC.ts.flag == 1)
  {
    sensor_msgs::TimeReference trigTime;
    trigTime.header.stamp = hardSync_FC.ts.time2p5ms * 2.5ms; // won't work as it is for type reason, but you get the idea
    trigTime.time_ref     = now_time;
    trigTime.source       = "FC";

    p->trigger_publisher.publish(trigTime);
  }
```

But it won't work for two reasons: 
- trigTime.header.stamp is of type ros::time (as I pointed out in comment)
- FC's timestamps are computed from when FC started up while ROS' time are actual time

So, is it a mistake ?

dji-dev, 2019-11-14:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29836'>Zendesk ticket #29836</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The question needs further analysis and testing, if need further communication or there is any result, we will contact you immediately.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2019-12-18:
@dji-dev 
Do you have any news on this subject ?

dji-dev, 2019-12-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29836'>Zendesk ticket #29836</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

This problem has been upgraded to the relevant engineers to analysis, and it will take some time because of the task scheduling. Thanks again for your patience, and we will info you as soon as there is any result.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-07-21:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29836'>Zendesk ticket #29836</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry to keep you waiting, but the logic is fine, after you set align_time to true in the launch file, synchronize the data with FC before publishing, and base_time is available if (hardsync_fc.ts.flag == 1) the data happens to be emitted when the pulse is triggered. This data can be used to align the TIMING of ROS and FC. The MSG_time and Now_time in Trigger_time are identical at this point.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle