dji-sdk/Onboard-SDK-ROS #311: can not success to catkin_make

liu978079474, 2019-11-19:
Hello, catkin_make failed.I use Manifold 2

Ubuntu 16.04.2
ROS kinetic
Error log:
collect2: error: ld returned 1 exit status
sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/build.make:137: recipe for target 'bin/djiosdk-cameragimbal-sample' failed
make[2]: *** [bin/djiosdk-cameragimbal-sample] Error 1
CMakeFiles/Makefile2:224: recipe for target 'sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/all' failed
make[1]: *** [sample/linux/camera-gimbal/CMakeFiles/djiosdk-cameragimbal-sample.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
CMakeFiles/Onboard-SDK.dir/build.make:73: recipe for target 'Onboard-SDK-prefix/src/Onboard-SDK-stamp/Onboard-SDK-install' failed
make[2]: *** [Onboard-SDK-prefix/src/Onboard-SDK-stamp/Onboard-SDK-install] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Onboard-SDK.dir/all' failed
make[1]: *** [CMakeFiles/Onboard-SDK.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
CMake Error at Onboard-SDK-ROS-3.8/dji_sdk/CMakeLists.txt:31 (find_package):
  By not providing "FindDJIOSDK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DJIOSDK", but
  CMake did not find one.

  Could not find a package configuration file provided by "DJIOSDK" with any
  of the following names:

    DJIOSDKConfig.cmake
    djiosdk-config.cmake

  Add the installation prefix of "DJIOSDK" to CMAKE_PREFIX_PATH or set
  "DJIOSDK_DIR" to a directory containing one of the above files.  If
  "DJIOSDK" provides a separate development package or SDK, be sure it has
  been installed.

dji-dev, 2019-11-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29992'>Zendesk ticket #29992</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you compile according to the steps indicated in the DJI OSDK development documentation, https://developer.dji.com/onboard-sdk/documentation/development-workflow/sample-setup.html#ros-onboard-computer
If a compilation error still occurs, it is recommended to provide a complete compilation log (including all commands executed).

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

liu978079474, 2019-11-20:
Tanke you for your reply!But I still have catkin_make unsuccessful.
log:
dji@manifold2:~/catkin_ws$ catkin_make
Base path: /home/dji/catkin_ws
Source space: /home/dji/catkin_ws/src
Build space: /home/dji/catkin_ws/build
Devel space: /home/dji/catkin_ws/devel
Install space: /home/dji/catkin_ws/install
####
#### Running command: "cmake /home/dji/catkin_ws/src -DCATKIN_DEVEL_PREFIX=/home/dji/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/dji/catkin_ws/install -G Unix Makefiles" in "/home/dji/catkin_ws/build"
####
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Using CATKIN_DEVEL_PREFIX: /home/dji/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/dji/velodyne/devel;/home/dji/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/dji/velodyne/devel;/home/dji/catkin_ws/devel;/opt/ros/kinetic
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/dji/catkin_ws/build/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.18
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 2 packages in topological order:
-- ~~  - dji_sdk
-- ~~  - dji_sdk_demo
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'dji_sdk'
-- ==> add_subdirectory(Onboard-SDK-ROS-3.8/dji_sdk)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Using Finddjiosdk-core.cmake find djiosdk-core
-- DO NOT FIND THE djiosdk-core LIB, WILL TRY TO DOWNLOAD FORM GITHUB, COMPILE, INSTALL
-- The C compiler identification is GNU 5.5.0
-- The CXX compiler identification is GNU 5.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
[... 240 lines cut ...]
make[2]: *** [Onboard-SDK-prefix/src/Onboard-SDK-stamp/Onboard-SDK-install] Error 2
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/Onboard-SDK.dir/all' failed
make[1]: *** [CMakeFiles/Onboard-SDK.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
CMake Error at Onboard-SDK-ROS-3.8/dji_sdk/CMakeLists.txt:31 (find_package):
  By not providing "FindDJIOSDK.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "DJIOSDK", but
  CMake did not find one.

  Could not find a package configuration file provided by "DJIOSDK" with any
  of the following names:

    DJIOSDKConfig.cmake
    djiosdk-config.cmake

  Add the installation prefix of "DJIOSDK" to CMAKE_PREFIX_PATH or set
  "DJIOSDK_DIR" to a directory containing one of the above files.  If
  "DJIOSDK" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/dji/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

dji-dev, 2019-11-21:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/29992'>Zendesk ticket #29992</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Noticed that the error occurred when compiling and link djiosdk-cameragimbal-sample, so what version of FFMPEG are you using? It is recommended to install 2.8.15, and if the compilation error still occurs, it is recommended that you restore manifold2-G's default system.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle