dji-sdk/Onboard-SDK-ROS #312: Want to use onboard ROS in Matrice 100

sagardalai, 2019-11-19:
Hello,

I am new to this, so i want use onBoard ROS in matrice 100.

First of all I have sort of queries.
1. Does the flight controller of matrice 100 has Linux os in bulid.
if yes then
2. does it has been pre-installed ROS or i need to do that.

if there is no OS (LInux) installed on it then 
1. How can I run my ROS code to Flight the Matrice 100.

If i am in a wrong track then please guide me.

Thanks in advance.

pavankumarbn, 2019-11-19:
@sagardalai You need to use the Onboard computer to roll on.

sagardalai, 2019-11-19:
@pavankumarbn  actually i need to run my own code on matrice 100.

erenaud3, 2019-11-19:
Okay, so the first thing you must keep in mind when working with a DJI drone : **you don't have any kind of control of what is inside.**

So for question 1. : nobody except DJI can tell what is actually on the drone (and they won't tell you). But it won't be a Linux system though. Most certainly, they have their own firmware on their own flight controller (just like you would program an Arduino for example).
So answer for question 2. is obviously no, they isn't any pre-installed ROS.

What you failed to understand is that you need to add a "computer" to your Matrice 100, which is called "on-board computer" on documentation (or just OC). This computer will be connected to the flight controller (FC) thanks to a classical Rx-Tx line. It could be anything (as long as it is ROS capable), from Raspberry to a real computer. You should definitely read [this tutorial](http://wiki.ros.org/dji_sdk/Tutorials/Getting%20Started). Everything will be clearer after that.

So once you have chosen an OC and installed ROS on it, what you can do is listen to the data sent by the Matrice 100 and sent some pre-defined commands. But I insist, you do not control anything on the drone. You can only use what DJI's developers provide for you.

> actually i need to run my own code on matrice 100.

So as I just explained, you can't do that. What did you have in mind ? Maybe you can find some workaround using commands from SDK.

> How can I run my ROS code to Flight the Matrice 100.

Trust me, you don't want do to that. It's probably the best way to crash your drone. Or at least not in real life. But you can learn how to perform a simulation (tutorial above).

More generally:

- If you want full control over your drone, what you are looking for is some kind of open-source solution (based on open-source flight controller, such as PixHawk). It probably won't fly as good as DJI's drones though.
- I suspect that you didn't read so much documentation on [dji_sdk ROS package](http://wiki.ros.org/dji_sdk) nor [OnBoard-SDK](https://developer.dji.com/onboard-sdk/documentation/introduction/homepage.html). You definitely should. 

Good luck !

pavankumarbn, 2019-11-20:
@erenaud3 I appreciate your detailed explanation. It really helps @sagardalai.

sagardalai, 2019-11-22:
Thank you @erenaud3  and @pavankumarbn  for the information.
Hope I will succeed to overcome from this issue.

sagardalai, 2019-11-26:
Hello @erenaud3,

As you mentioned above 
1. I suspect that you didn't read so much documentation on dji_sdk ROS package nor OnBoard-SDK. You definitely should.
using onboard computer and dji sd ROS package can we control our drone through our code

pavankumarbn, 2019-11-27:
@sagardalai yes you can control. There are sample examples available, you can check with them.  (https://www.youtube.com/watch?v=2WMTu2o1iq4) It shows what you needed.

erenaud3, 2019-11-27:
> Hello @erenaud3,
> 
> As you mentioned above
> 
>     1. I suspect that you didn't read so much documentation on dji_sdk ROS package nor OnBoard-SDK. You definitely should.
>        using onboard computer and dji sd ROS package can we control our drone through our code

Did you see [this](http://wiki.ros.org/dji_sdk/Tutorials/Running%20the%20flight%20control%20sample%20code) ?

closed 2021-04-09 by sagardalai