dji-sdk/Onboard-SDK-ROS #313: Unable to 'catkin_make' for DJI_Onboard-SDK-ROS

neaylol, 2019-11-20:
M210 RTK V2
Ubuntu16.04 LTS
Manifold2

I tried to execute the 'catkin_make' command, but I get the below error.
usr/local/include/djiosdk/WaypointV2.pb.h:9:39:fatal error:dji/protobuf/stubs/common.h:No such file or directory

Please help me on how to proceed further.

Thanks

dji-dev, 2019-11-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30041'>Zendesk ticket #30041</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you compile according to the steps indicated in the DJI OSDK development documentation, https://developer.dji.com/onboard-sdk/documentation/development-workflow/sample-setup.html#ros-onboard-computer

If a compilation error still occurs, and in order to better understand and help you solve the problem, suggest you click the hyperlink on the right to fill in and submit more complete information about the problem.
https://formcrafts.com/a/dji-developer-feedback-en
1. The complete compilation log (Includes all commands you execute)
2. The version of OSDK-ROS?
3. DJI manifold2-G or -C version?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

pzy2253999075, 2019-12-06:
I meet the same error,have you @neaylol   solved the problem？ I use M210 V2 ，manifold2-G and the version of OSDK-ROS is 3.8.1.Could you help me @dji-dev

dji-kyle, 2021-07-08:
Please install the OSDK lib and .h file first. The OSDK-ROS based on the OSDK linux lib.

closed 2021-07-08 by dji-kyle