dji-sdk/Onboard-SDK-ROS #315: M210 V2    连接妙算2g  使用onboardsdk  接线文问题

pzy2253999075, 2019-11-27:
请问，m210的窜口怎么接到妙算上，我想要使用onboardsdk  ，还要开启advance sensing 
[image]

pzy2253999075, 2019-11-27:
@dji-dev 可以帮忙解决下嘛

dji-dev, 2019-11-28:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30247'>Zendesk ticket #30247</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

建议您可以参考DJI OSDK开发文档关于Hardware Setup，Software Environment Setup等部分内容。
https://developer.dji.com/cn/onboard-sdk/documentation/development-workflow/hardware-setup.html
https://developer.dji.com/cn/onboard-sdk/documentation/development-workflow/environment-setup.html

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

pzy2253999075, 2019-11-28:
@dji-dev   我按照那2个链接的流程做了，还是不行，报同样的错误。请问，首先我这个无人机到妙算端的连线正确吗，启动onboardsdk的时候应该不用打开遥控器和app吧

dji-dev, 2019-11-29:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30247'>Zendesk ticket #30247</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

已在您的邮件（Ticket 30262）回复，请注意查收。
另外第一运行OSDK是需要进行激活操作，建议打开遥控器并连接手机运行DJI Pilot。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

closed 2021-07-08 by dji-kyle