dji-sdk/Onboard-SDK-ROS #316: M210 V2 ，OSDK  can't not setup camera video stream

pzy2253999075, 2019-12-06:
I recently used the onboardsdk on the M210 and manifold2-G. I managed to activate the OSDK, and then I found another problem that I couldn't accept the video stream from the camera.After I roslaunch sdk.launch,I use the command "rosservice call /dji_sdk/setup_camera_stream cameraType:1 ,start: 1".I got this error:
[image]

How can I solve this problem?

Software and hardware list：
1、M210 v2
2、manifold2-G
3、Zenmuse Z30（camera）
4、OnboardSdk  （version 3.8）
5、OSDK_ROS (version 3.8.1)
6、firmware  version：1.0.0450

@dji-dev Could you help to fix this problem?Thanks

dji-dev, 2019-12-09:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30566'>Zendesk ticket #30566</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

May I ask whether it is normal to set the cameraType to 0 to get the FPV camera stream?
Judging from the log error message, it is likely to be related to the ROS environment, and it is recommended that you can try other environments (such as x86 ubuntu16.04) for testing and validation.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

pzy2253999075, 2019-12-10:
I'm using Ubuntu 16.04, which is the default system of manifold. And the ROS environment has been pre installed Already.
This is the output when I set the cameraType to 0 for FPV camera stream:
[image]

When I run the OSDK sample （camera-stream-poll-sample）in Onboard-Sdk-3.8/bulid/bin Dir，.
For main camera stream,I got this:
[image]
And For FVP camera stream, I got this:
[image]
But it stopped running for a while by itself，and  then I got the ending：
[image]

pzy2253999075, 2019-12-12:
@dji-dev Could you help me with that？

dji-dev, 2019-12-13:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30566'>Zendesk ticket #30566</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

It is recommended that you can use the OSDK-ROS Master branch for testing and validation, which fixes the problem of getting video streams from the main camera. https://github.com/dji-sdk/Onboard-SDK-ROS/tree/master

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

pzy2253999075, 2019-12-16:
@dji-dev Thanks for your answer.Now I change the software environment. 
I use:
1、OSDK-ROS   Master branch
2、OSDK   Master branch
3、firmware  1.0.0450

I can  successfully roslaunch sdk.launch like this below:
[image]

But I when I use the command  "rosservice call /dji_sdk/setup_camera_stream" and  set the cameraType to 1 to get main stream.I got this output:

[image]

But I can't watch the video in rqt_image_view.And when I rostopic echo /dji_sdk/main_camera_images,there is nothing output. I notic that the default baudrate for ttyACM0 is 230400,I don't know if it's correct.

So when I set the cameraType to 0 to get fpv stream.I got this:
[image]

and I can get the data when typing rostopic echo /dji_sdk/fpv_camera_images.So now there is still a problem in obtaining the mian stream. How to solve this problem？Could you help me?

dji-dev, 2019-12-17:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30566'>Zendesk ticket #30566</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that this problem cannot be reproduced after actual testing and verification. It is recommended that you can upgrade the firmware of the M210 V2 and Z30 to the latest version before verifying and testing. In addition, the baud rate of ttyACM0 is correct, it is not the same baud rate as the API interface.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle