dji-sdk/Onboard-SDK-ROS #318: about  OSDK zoom control interface

pzy2253999075, 2019-12-19:
@dji-dev Is there a camera zoom interface in OSDK? My Z30 has zoom function, but I don't know if OSDK provides a zoom control interface.

dji-dev, 2019-12-19:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30916'>Zendesk ticket #30916</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

DJI Onboard SDK support controls the zoom of Z30 Camera (Only support for Matrice 210 V2 and Matrice 210 RTK V2).
It is suggested that you can refer to the OSDK document, https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-camera-and-gimbal.html

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

pzy2253999075, 2019-12-24:
@dji-dev Thanks for your answer，but I didn't find the specific operation from doc you given.Could you tell me some commamd lines such as "rostopic pub **" or "rosservice call **" directly.I use this in ROS.I only found a related rossservice named "/dji_sdk/camera_action".But I don't know which value corresponds to zoom.I've looked for the camera control information in osdk, but I haven't found the zoom, for example
[image]
Could you tell me how to do? Thanks again.

dji-dev, 2019-12-26:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30916'>Zendesk ticket #30916</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

Sorry that the current version of OSK-ROS-3.8 does not directly support control of camera zoom, but you can find the sample code in OSDK-3.9 Linux, https://github.com/dji-sdk/Onboard-SDK/blob/master/sample/platform/linux/payloads/main_sync.cpp#L153

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle