dji-sdk/Onboard-SDK-ROS #319: rc data problem

oycq, 2019-12-20:
We have an A3 controler.
when we run `rostopic echo /dji_sdk/rc` , the rc data only echo for a short duration.(just one or two seconds when we run ` roslaunch dji_sdk sdk.launch`
what is the fllowing message means? [ INFO] [1576822311.346397705]: align_time_with_FC set to false. We will use ros time to time stamp messages!
By the way my telephone is 18408222921. I am a chinese .:.>

... logging to /home/oycq/.ros/log/ad4925e2-22ed-11ea-8846-0c9d92c42524/roslaunch-titan-10852.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://titan:32953/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: 1080161
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 230400
 * /dji_sdk/enc_key: e362a40fba4052397...
 * /dji_sdk/serial_name: /dev/dji_uart
 * /dji_sdk/use_broadcast: False
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

ROS_MASTER_URI=http://localhost:11311

process[dji_sdk-1]: started with pid [10867]

STATUS/1 @ init, L56: Attempting to open device /dev/dji_uart with baudrate 230400...

STATUS/1 @ init, L66: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L762: Device Serial No. = 0670153997

STATUS/1 @ parseDroneVersionInfo, L764: Hardware = A3

STATUS/1 @ parseDroneVersionInfo, L765: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1188: Virtual RC is not supported on this platform!

STATUS/1 @ activate, L1398: version 0x303082F

STATUS/1 @ activate, L1411: Activation successful

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ initGimbal, L926: Checking if gimbal is connected ...
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=50, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ initGimbal, L970: Gimbal not mounted!
[ INFO] [1576822311.308268753]: drone activated
[ INFO] [1576822311.346342157]: Use data subscription to get telemetry data!
[ INFO] [1576822311.346397705]: align_time_with_FC set to false. We will use ros time to time stamp messages!

STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: 
STATUS/1 @ startPackage, L355: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L353: 
STATUS/1 @ startPackage, L355: Package 1 info: freq=50, nTopics=14.

STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=5, nTopics=6.

STATUS/1 @ startPackage, L353: 
STATUS/1 @ startPackage, L355: Package 3 info: freq=400, nTopics=1.

dji-dev, 2019-12-20:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/30953'>Zendesk ticket #30953</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to the information you provided, we cannot determine the cause of the problem. In order to better understand and help you solve the problem, recommend that you click the hyperlink on the right to fill in more detailed problem information,
https://formcrafts.com/a/dji-developer-feedback-en

what is the fllowing message means? [INFO] [1576822311.346397705]: align_time_with_FC set to false. We will use ros time to time stamp messages!
==> This is to remind you that the current system timestamp uses the time of the ROS system.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

oycq, 2019-12-20:
.... I have solved this problem. Just change baud_rate from 23400 to 921600.

closed 2019-12-20 by oycq

zhanguo33, 2020-10-31:
博主您好
我现在a3飞控是1.7.70，用的是osdk-ros是3.8的版本，波特率改了961200还是不好用，求解。