dji-sdk/Onboard-SDK-ROS #321: dji_sdk_demo/src/demo_flight_control.cpp, NED or ENU?

waterretaw, 2020-01-09:
在 dji_sdk_demo/src/demo_flight_control.cpp 文件中
```c++
/*! Very simple calculation of local NED offset between two pairs of GPS
/coordinates. Accurate when distances are small.
!*/
void
localOffsetFromGpsOffset(geometry_msgs::Vector3&  deltaNed,
                         sensor_msgs::NavSatFix& target,
                         sensor_msgs::NavSatFix& origin)
{
  double deltaLon = target.longitude - origin.longitude;
  double deltaLat = target.latitude - origin.latitude;

  deltaNed.y = deltaLat * deg2rad * C_EARTH;
  deltaNed.x = deltaLon * deg2rad * C_EARTH * cos(deg2rad*target.latitude);
  deltaNed.z = target.altitude - origin.altitude;
}
```
在 NED 坐标系中:
x--N--latitude，x应该是维度的改变量
y--E--longitude， y应该是经度的改变量
但是代码里：
deltaNed.y = deltaLat ...
deltaNed.x = deltaLon ...
这个代码是不是计算的ENU坐标系下的偏移量？

dji-dev, 2020-01-10:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31435'>Zendesk ticket #31435</a>:<br><br>尊敬的客户，

您好！感谢您联系DJI大疆创新。

该问题需要做进一步的验证和确认，一有结果我们将第一时间通知您。

感谢您的理解与支持,祝您生活愉快!

Best Regards，
DJI大疆创新SDK技术支持

dji-kyle, 2021-07-08:
OSDK-ROS使用的是ROS REP 103 标准，使用的是ENU坐标系，是计算ENU坐标系下的偏移量。程序注释有误。

closed 2021-07-08 by dji-kyle