dji-sdk/Onboard-SDK-ROS #322: DJI Manifold2C to A3 connection

arixrobotics, 2020-01-29:
I'm trying to connect the DJI Manifold2C to the A3 on my M600 Pro.
I've connected the cable from A3's API port, and the UART0 of the Manifold2C.
The problem is, I don't think the Manifold detected the A3.  
```
dji@dji:~$ ls /dev/ttyUSB*
ls: cannot access '/dev/ttyUSB*': No such file or directory
```

Looking at the Manifold2's user guide, it mentions that the cable should be connected to the UART**1** port of the Manifold2**G**. Unfortunately, there is no mention on how to connect to Manifold2**C**, since there is only UART**0**.

Can you show me how to connect in this configuration?

dji-dev, 2020-01-30:
Public comment from Lisa Fedane in <a href='https://djisdksupport.zendesk.com/agent/tickets/31856'>Zendesk ticket #31856</a>:<br><br>DJI engineers will be on holiday for the Chinese Spring Festival (1-23 ~ 02-9), which may cause some delays in responding to your ticket. We appreciate your patience and apologize for the inconvenience this may cause you.

Lisa Fedane
DJI Developer Support

dji-dev, 2020-02-03:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/31856'>Zendesk ticket #31856</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

The DJI Manifold2-C does not have UART1, so it is recommended that you can use external serial port devices, such as USB-TTL, and FT232.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle