dji-sdk/Onboard-SDK-ROS #324: Failed sending mission start command

Xachap, 2020-02-21:
Hi,

I have a problem when trying to run "rosrun dji_demo demo_mission", the program starts correctly but when it tryes to start one of the two available modes (waypoint or hotpoint mission) it fails with the following error:

`[ WARN] [1582271165.048665289]: Failed sending mission start command`

Its quite strange because I have the simulator running at the same time, and all works fine if I call directly to a service, like for example: 
`rosservice call /dji_sdk/drone_task_control "task: 4"`

If I select waypoint mission, I get this trace:
`user@ubuntu:~/catkin_ws$ rosrun dji_sdk_demo demo_mission 
[ INFO] [1582271103.707979851]: Activated successfully
[ INFO] [1582271113.711974457]: Obtain SDK control Authority successfully
| Available commands:                                            |
| [a] Waypoint Mission                                           |
| [b] Hotpoint Mission                                           |
a
[ INFO] [1582271116.742298632]: Creating Waypoints..

[ INFO] [1582271116.742430330]: Waypoint created at (LLA): 22.542813 	113.958902 	10.000000

[ INFO] [1582271116.742556372]: Uploading Waypoints..

[ INFO] [1582271116.742654840]: Waypoint created at (LLA): 22.542813 	113.958902 	10.000000
 
[ INFO] [1582271116.742685412]: Waypoint created at (LLA): 22.542842 	113.958952 	11.000000
 
[ INFO] [1582271116.742746459]: Waypoint created at (LLA): 22.542813 	113.959001 	12.000000
 
[ INFO] [1582271116.742851195]: Waypoint created at (LLA): 22.542756 	113.959001 	13.000000
 
[ INFO] [1582271116.742901448]: Waypoint created at (LLA): 22.542727 	113.958952 	14.000000
 
[ INFO] [1582271116.742959309]: Waypoint created at (LLA): 22.542756 	113.958902 	15.000000
 
[ INFO] [1582271116.743012472]: Waypoint created at (LLA): 22.542813 	113.958902 	10.000000
 
[ INFO] [1582271116.743062476]: Initializing Waypoint Mission..

[ INFO] [1582271164.991558204]: Waypoint upload command sent successfully
[ WARN] [1582271165.048617001]: ack.info: set = 3 id = 18
[ WARN] [1582271165.048644037]: ack.data: 205
[ WARN] [1582271165.048665289]: Failed sending mission start command`

And this is for hotpoint mission:
`user@ubuntu:~/catkin_ws$ rosrun dji_sdk_demo demo_mission 
[ INFO] [1582271190.690204204]: Activated successfully
[ INFO] [1582271190.749468179]: Obtain SDK control Authority successfully
| Available commands:                                            |
| [a] Waypoint Mission                                           |
| [b] Hotpoint Mission                                           |
b
[ INFO] [1582271221.395103953]: Takeoff command sent successfully
[ INFO] [1582271236.395674023]: Start with default rotation rate: 15 deg/s
[ WARN] [1582271236.453233603]: ack.info: set = 3 id = 32
[ WARN] [1582271236.453291867]: ack.data: 205
[ WARN] [1582271236.453362642]: Failed sending mission start command
`

Thank you very much in advanced! :)

dji-dev, 2020-02-24:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32291'>Zendesk ticket #32291</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to the information you provided, we are unable to determine the cause of the problem. In order to better understand and help you solve the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary ROS main node log/screenshot/video), https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-kyle, 2021-07-08:
Please update to the latest firmvare version and OSDK version, verify this issue with sample

closed 2021-07-08 by dji-kyle