dji-sdk/Onboard-SDK-ROS #325: Changing IMU publishing rate

maxhofidrone, 2020-02-28:
Hi all,
I am planing to do a sensor fusion of the `/dji_sdk/imu` and `/dji_sdk/rtk_position` topics as basis for a future flight controller for my DJI M600 Pro RTK.
By analyzing the IMU data, I recognized that the publishing frequency is quite varying with time. These variation have some kind of specific repetition pattern: 
[image]
For this plot i just calculated the difference of the ros time information of the ros header of every `/dji_sdk/imu` message (I recorded data around one hour, leaving the M600 on the ground / Matlab: `diff(Header.secs+Header.nsecs*10^-9)`). The average of all the differences is very close to 0.0025 seconds which also relates to a publishing rate of 400Hz. But in reality there are close to non messages published with this rate.

What could be the reason of this behavior? I'm using the current firmware (3.2.41.14) and the current OSDK-ROS. As the same behavior appears on different "onboard" PCs (Raspberry Pi 4 &  a Intel® Core™ i7-9700K CPU @ 3.60GHz × 8 32BG Dell) I am sure it is not the fault of the onboard computer (internal clock, CPU load, USB buffer things...). Does anyone had this behavior too? If so how do you  handle it, especially in terms of a sensor fusion where  at least a stable IMU rate is necessary for the strap down calculation? 

Looking forward for any help!
All the best, 
Max

dji-dev, 2020-03-02:
Public comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/32466'>Zendesk ticket #32466</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to your feedback, this is probably the reason for the hardware connection and the setting. It is recommended that you can use the USB-TTL device of FT232 and the serial port baud rate of 1M. The connected TTL line should be as short as possible/with good contact to avoid electromagnetic interference, and then record a rosbag for analysis.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

erenaud3, 2020-03-05:
Dear @maxhofidrone,

I think the problem comes from ROS. Indeed, if you take a look at this function:
https://github.com/dji-sdk/Onboard-SDK-ROS/blob/fe32f40ae008ba62a668875c27ca5a312e62601b/dji_sdk/src/modules/dji_sdk_node_publisher.cpp#L630-L701

You will notice that DJI is not using Flight Controller (FC) time to timestamps their message (packageTimeStamp is basically not used. Unless you activate time alignment: p->align_time_with_FC == true). Instead, they use the ROS time at which your Onboard Computer (OC) received IMU package.

As we connect OC to FC using UART protocol, data exchange is asynchronous. So publish rate can't be constant by definition, it will depends on what your OC is doing at that time. That's why you are seeing IMU rate being inconsistent.

**Salty mode enabled**

My team just decided to drop out DJI M600. Why ? Because they sell you a product described as "open" and you end up with something so closed you can't do anything. Plus we heard from our drone retailer that DJI is planning on dropping out M600 by the end of the year (cause legislation is evoloving in a way such that M600 won't be allowed to fly as it currently is). So yeah, really, you don't want to keep using that one. 

We are planning on moving out to open source solution (PX4).

maxhofidrone, 2020-03-26:
Dear @dji-dev ,
thanks for your help. You are right this issue is related to the buffering of data in the serial USB connection. Unfortunately this behavior can not be eliminated. 

Dear @erenaud3,
 thank you very much for your elaboration. It is quite sad that DJI is giving us users such a hard time... Did you already decided for PX4 platform? If you like we may could get in contact about this?

closed 2020-03-26 by maxhofidrone

erenaud3, 2020-03-27:
@maxhofidrone, yeah I started playing around with a Pixhawk 1. Really interesting and lot of potentials.

maxhofidrone, 2020-03-27:
@erenaud3  Thanks! I checked it also a bit already. Did you also already decided to an airframe? I guess you need also a platform with similar payload as the M600.

erenaud3, 2020-03-27:
@maxhofidrone We do not plan on building the drone ourselves. I would have love to but we simply don't have the time nor the human resources. Instead, we will buy a functional one based on Pixhawk 4 and PX4 (this [retailer](http://innovadrone.com/), but french website only sorry).

I think that's really the strength of PX4: as it is under BSD licence, lot of drone manufacturers use it. So once you prototype something with your own standalone board, it shouldn't be too hard to make it work with any PX4 drone.

maxhofidrone, 2020-03-30:
@erenaud3 merci! Good luck with your project!