dji-sdk/Onboard-SDK-ROS #326: Could I use DJI M100 or DJI M210 fully autonomous based on OSDK without remote controller

frontw, 2020-03-10:
Hi!

Is it possible to use DJI M100 or DJI M210 for tasks with full automation, where no operator and remote controller are present?
By practice, I couldnt't aquire OSDK control when remote controller is turned off. Is it possible to avoid it?

erenaud3, 2020-03-11:
No. And the reason is legal: you can't have a drone flying automatically without any mean to take control back. What if your software/hardware is malfunctioning and your drone is going straight to someone's head ?

frontw, 2020-03-11:
> No. And the reason is legal: you can't have a drone flying automatically without any mean to take control back. What if your software/hardware is malfunctioning and your drone is going straight to someone's head ?

Actually, I received another reply

> The DJI Onboard SDK supports controlling M100 or M210 without remote control but requires personnel to manually turn on the power switch of the aircraft.
> You can refer to the relevant Settings, https://developer.dji.com/onboard-sdk/documentation/guides/component-guide-control-authority.html#remote-controller-lost
> 

Need to test.
If it's possible, then I could use DJI drone with autonomous charging station, elsewhere I should use PX4, which has no limits.

erenaud3, 2020-03-12:
That's actually interesting !

> If it's possible, then I could use DJI drone with autonomous charging station,

Are you talking about [this](https://www.heishatech.com/charging-pad-3/) ? 

But as I said, what you want to achieve may be illegal depending on which country you live in. For example, it is in France (and all European countries as well I think).

frontw, 2020-03-12:
> > If it's possible, then I could use DJI drone with autonomous charging station,
> 
> Are you talking about [this](https://www.heishatech.com/charging-pad-3/) ?

Yes, here is ours https://youtu.be/dAs48rKtT5o

closed 2021-07-08 by dji-kyle