dji-sdk/Onboard-SDK-ROS #330: WAYPOINT_MISSION_CHECK_FAILED

SpencerHehl, 2020-04-07:
I'm running the Onboard-SDK-ROS with an A3 Pro and while uploading a 2 waypoint mission received a WAYPOINT_MISSION_CHECK_FAILED. I have been unable to find any information regarding this error in the documentation that provides insight into what exactly is being checked and why it is failing.

dji-dev, 2020-04-08:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/33387'>Zendesk ticket #33387</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

According to the information you provided, we are unable to determine the cause of the problem. In order to better understand and help you solve the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary log/screenshot/video), https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

SpencerHehl, 2020-04-08:
I've submitted, but it is quite a simple question: what does WAYPOINT_MISSION_CHECK_FAILED even mean. The closest thing I can find is under https://developer.dji.com/onboard-sdk/documentation/M100-Docs/application-development-guides/flight-mission-ack-codes.html but that just says "check failed". What check was performed? Why does it fail? How do I troubleshoot this?

dji-dev, 2020-04-09:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/33387'>Zendesk ticket #33387</a>:<br><br>Dear Customer,

Thank you for contacting DJI.

 Sorry that we have not received the problem form you provided. May I ask what is the email account you filled in the form?
We need to know where the error occurred in your code so that we can better determine what causes the failure that OSDK is checking.
So we need more detailed information about the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary log/screenshot/video), https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

SpencerHehl, 2020-04-09:
I'm working with a support representative via email after submitting a ticket using that link already. However, since I would imagine others may encounter the error, I think it would be helpful to make sure relevant information is posted here. The above specified DJI error happens upon attempting to upload a mission, and appears to be a standard error output which is why I was just hoping it should be a simple answer as to what throws that error on the DJI side since it is lacking in really any descriptive information (unlike other errors such as waypoints too far).

dji-kyle, 2021-07-08:
Please use A3 + OSDK 3.9 , if there is any problems, please reopen again.the WAYPOINT_MISSION_CHECK_FAILED error is most likely the second time you uploaded fewer than 2 waypoints.

closed 2021-07-08 by dji-kyle