dji-sdk/Onboard-SDK-ROS #331: M210 change gimbal mode with Gimbal.msg

rvannamen, 2020-04-30:
Hi, 

I have a question regarding gimbal control with the DJI M210 V2. During flight I want to switch back and forth between the gimbal modes 'free' and 'follow'. I know this should by possible by using the Mobile SDK (version latest), however I would like to use the ROS Onboard SDK (version 3.8) for this. 

When controlling the gimbal by publishing on the topic /dji_sdk/gimbal_angle_cmd, there is an entry in the dji_sdk/Gimbal message where you can specify the mode. By setting this value to 1 or 0, you can switch between absolute en relative control. The message definition (https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/msg/Gimbal.msg) tells me that I can use other bytes to switch between 'free' and 'follow' mode as well. However, when a provide a value other than 0 or 1, the gimbal doesn't move at all, so I can not get the switching between 'free' and 'follow' to work. 

Could you provide any insights?

dji-dev, 2020-04-30:
Agent comment from Hummels Lei in <a href='https://djisdksupport.zendesk.com/agent/tickets/34109'>Zendesk ticket #34109</a>:<br><br>Hi,

You say "The message definition (https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/msg/Gimbal.msg) tells me that I can use other bytes to switch between 'free' and 'follow' mode as well". I am sorry.  I did n't see such a description. Could you please point it out for me?

dji-dev, 2020-05-06:
Agent comment from Bruce Cheung in <a href='https://djisdksupport.zendesk.com/agent/tickets/34109'>Zendesk ticket #34109</a>:<br><br>Dear customer,

Thank you for contacting DJI.

I can see your point, but according to the information you provided, we are unable to determine the cause of the problem. In order to better understand and help you solve the problem, it is recommended that you click the hyperlink on the right to fill in more detailed problem information (including the necessary log/screenshot/video, DJI Gimbal model, Gimbal mount location, Actual flight test or Simulation test?), https://formcrafts.com/a/dji-developer-feedback-en

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-08 by dji-kyle