dji-sdk/Onboard-SDK-ROS #332: How to get image data with OSDK using DJI matrice 600 pro

yyf202xz, 2020-05-15:
I'm trying to do some image-processing using matrice 600pro with a jetson Xavier attached.
A camera which has HDMI output function is attached to ronin-MX(gimbal) and data will be transmitted through SRW-60G(wireless video link using hdmi port).
I thought using some functions in onboard sdk such like "/dji_sdk/main_camera_images (sensor_msgs/Image)" (http://wiki.ros.org/dji_sdk) can get image data easily,
but I found that those functions are only available for M210 so I may cannot use these on my matrice 600pro.
Using a HDMI-USB converter may could solve this problem(make the camera-transmitter-receiver as a usb camera)
but the converter is quite expensive and I'm not sure if there's better way to do this.
Any clue will be very helpful. Thank you!

dji-dev, 2020-05-29:
Dear Customer,

Thank you for contacting DJI.

Sorry that the DJI OSDK-ROS gets the video stream only supports the M210 series, not M600 Pro, and does not support USB camera (even if using HDMI-USB converter).
Since you can get the HDMI/wireless video stream for a custom camera, it is recommended that you can create your own video streaming drivers and related ROS topics.

yyf202xz, 2020-06-03:
@dji-dev Thank you for reply. After I attached a capture card to the output side of videolink(SRW-60G), I got video stream from Sony handcam on my onboard computer finelly. However, there's a problem about SRW-60G, can I ask you here? When I change the handcam to gopro, there's no signal anymore. SRW-60G shows "Linking but the Transmitter is not connected to the HDMI device". The gopro was surelly in a hdmi output mode and the capture card works when input is a handcam. So I think maybe SRW-60G did a lille bit change to the signal from gopro that make it unreadble for the capturecard. Can you tell me some details about SRW-60G or someone else I can ask?

yyf202xz, 2020-06-18:
↑Attaching a HDMI splitter solved this problem! I don't know why but it works!

closed 2020-06-18 by yyf202xz