dji-sdk/Onboard-SDK-ROS #333: Is there's 3.9 version of Onboard-SDK-ROS?

yyf202xz, 2020-05-15:
I'm using Matrice 600pro and found the newest OSDK that compatible with it is 3.9(If I didn't get it wrong). After install the linux package and forward to install the ros package, there's no 3.9 in github branches(the newest version is 3.8). So are these packages don't need to be same at all?(e.g. OSDK:3.9 with OSDK-ros:3.8) Or its uploaded in another website? I'm sorry if I missed some message that cause this stupid question. Thank you.

mhoejgaard, 2020-05-18:
AFAIK, unfortunately, there is not and will not be a OSDK-ROS:3.9 release.

yyf202xz, 2020-05-19:
@mhoejgaard Thank you for reply! So can I use OSDK-ros:3.8 with OSDK:3.9? Are they compatible?

mhoejgaard, 2020-05-19:
@yyf202xz We have tested that without succes. Some functionality may work but not everything. The solution is probably to extend the ros wrapper to 3.9 yourself.

yyf202xz, 2020-05-19:
@mhoejgaard Thank you for the infomation. I will downgrade OSDK to 3.8 just to be sure every thing can work.

erenaud3, 2020-06-25:
> I'm using Matrice 600pro and found the newest OSDK that compatible with it is 3.9(If I didn't get it wrong). 

Hello @yyf202xz ,

May I ask where did you find this information ?

Thank you

yyf202xz, 2020-06-25:
> > I'm using Matrice 600pro and found the newest OSDK that compatible with it is 3.9(If I didn't get it wrong).
> 
> Hello @yyf202xz ,
> 
> May I ask where did you find this information ?
> 
> Thank you

Hello @erenaud3 ,
I checked each new branchs of OSDK on github. It's written in readme.md. For example this:
https://github.com/dji-sdk/Onboard-SDK/tree/3.9

closed 2021-07-08 by dji-kyle