dji-sdk/Onboard-SDK-ROS #335: Process had died Exit Code -11 on Ubuntu 18.04, Jetson TX2

CrashxZ, 2020-05-22:
I installed the Onboard SDK 4.0 and I am getting an error (Exit code 11, process has died).
please note that this problem didn't exist when I had the older version of the SDK. 

Log : 

> nvidia@nvidia-desktop:~/dji_sdk_ros$ roslaunch dji_osdk_ros dji_vehicle_node.launch 
> ... logging to /home/nvidia/.ros/log/2f736392-9c0c-11ea-b50b-00044bc6e804/roslaunch-nvidia-desktop-30840.log
> Checking log directory for disk usage. This may take a while.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://nvidia-desktop:39461/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /rosdistro: melodic
>  * /rosversion: 1.14.5
>  * /vehicle_node/acm_name: /dev/ttyACM0
>  * /vehicle_node/align_time: False
>  * /vehicle_node/app_id: 1075222
>  * /vehicle_node/app_version: 1
>  * /vehicle_node/baud_rate: 921600
>  * /vehicle_node/enc_key: 24054c76e33d17eb5...
>  * /vehicle_node/serial_name: /dev/ttyUSB0
>  * /vehicle_node/use_broadcast: False
> 
> NODES
>   /
>     vehicle_node (dji_osdk_ros/dji_vehicle_node)
> 
> auto-starting new master
> process[master]: started with pid [30853]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 2f736392-9c0c-11ea-b50b-00044bc6e804
> process[rosout-1]: started with pid [30864]
> started core service [/rosout]
> process[vehicle_node-2]: started with pid [30871]
> EnableAd: 1
> [1000776.683]ERRORLOG/1 @ initVehicle, L181: Failed to initialize ACM Linker channel!
> [1000776.687]STATUS/1 @ parseDroneVersionInfo, L1157: Device Serial No. = 0DBDG5F002009D
> [1000776.687]STATUS/1 @ parseDroneVersionInfo, L1159: Firmware = 3.3.8.47
> [1000776.688]STATUS/1 @ functionalSetUp, L315: Shake hand with drone successfully by getting drone version.
> [1000776.688]ERRORLOG/1 @ functionalSetUp, L334: Only support M210 V2 series and M300!
> 
> [1000776.688]STATUS/1 @ activate, L1268: version 0x303082F
> 
> [vehicle_node-2] process has died [pid 30871, exit code -11, cmd /home/nvidia/dji_sdk_ros/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/home/nvidia/.ros/log/2f736392-9c0c-11ea-b50b-00044bc6e804/vehicle_node-2.log].
> log file: /home/nvidia/.ros/log/2f736392-9c0c-11ea-b50b-00044bc6e804/vehicle_node-2*.log

There is no log file named "/home/nvidia/.ros/log/2f736392-9c0c-11ea-b50b-00044bc6e804/vehicle_node-2.log"

I am using a Jetson TX2 with kernel  4.9.140-tegra, trying to connect it to the DJI N3.
As mentioned earlier, this configuration worked earlier with the older version of the SDK

jeffav, 2020-05-22:
Are you using the M210 v2 or the M300? See this error in your output:
[1000776.688]ERRORLOG/1 @ functionalSetUp, L334: Only support M210 V2 series and M300!

CrashxZ, 2020-05-22:
Its a custom hexacopter frame, the flight controller is N3.  The DJI Onboard SDK and the DJI Onboard SDK ROS supports n3 according to the documentation and it works with the SDK version 3.8.  (I have it on the another TX2)

"Only support M210 V2 series and M300!" 
this is only for the video feed I presume.

jeffav, 2020-05-22:
I would verify that you can run one of the OSDK samples before testing the ROS layer, but the [OSDK ReadMe](https://github.com/dji-sdk/Onboard-SDK/blob/master/README.md) states that "OSDK 4.0 was released on 7 May 2020. This version supports M300 and M210/M210 RTK V2, you can try OSDK functions at M300 now."

CrashxZ, 2020-05-22:
I guess you're right, but I'm not sure why they would remove support for it. 
But it shows the supported products for the Onboard SDK here: https://developer.dji.com/onboard-sdk/
[image]

CrashxZ, 2020-05-22:
I will verify it on Monday, and report back.
 Thank You.

CrashxZ, 2020-05-22:
Also, the error was same when launching the sdk_node.launch (which is the old interface in the new wrapper)

dji-dev, 2020-05-29:
Dear Customer,

Thank you for contacting DJI.

We are sorry for the inconvenience caused to you. Currently, the DJI OSDK-ROS 4.0 only supports M210/RTK V2 and M300 RTK, and N3 is not supported for the time being. It is suggested that you can use version 3.8 of OSDK-ROS.

closed 2021-07-08 by dji-kyle