dji-sdk/Onboard-SDK-ROS #342: catkin build and install fails

jeffav, 2020-06-05:
When configuring catkin build to install
```
cd ~/catkin_ws
catkin config --install
catkin build
```
the install fails with this error
```
Errors     << dji_osdk_ros:install /home/developer/catkin_ws/logs/dji_osdk_ros/build.install.002.log                                                                                     
CMake Error at src/dji_osdk_ros_obsoleted/cmake_install.cmake:41 (file):
  file INSTALL cannot find
  "/home/developer/catkin_ws/src/Onboard-SDK-ROS/src/dji_osdk_ros_obsoleted/include/dji_sdk".
Call Stack (most recent call first):
  src/cmake_install.cmake:43 (include)
  cmake_install.cmake:253 (include)

make: *** [install] Error 1
```

dji-dev, 2020-06-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35472'>Zendesk ticket #35472</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About to prepare Development Environment, you can go to the DJI dev website for more:
https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html

Hope the above answers are helpful to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

jeffav, 2020-06-05:
To be clear, I am able to build the software using `catkin build` normally. But when I set the INSTALL flag using catkin config --install then run catkin build, it fails.

I believe there were some install commands in the CMake files that referred to old files before the 4.0 release. My pull request #343 fixes the issue.

dji-dev, 2020-06-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35472'>Zendesk ticket #35472</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

So sorry that the mistakes occured with cmd install,
It referrences to wrong path when install the files, you can change the cmake file as follow, if necessary. And we will forwared relative engineer to check and update into ROS-SDK。
\catkin_ws\build\Onboard-SDK-ROS-4.0\src\dji_osdk_ros_obsoleted\cmake_install.cmake
line36:
file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include/dji_osdk_ros" TYPE DIRECTORY FILES "/home/work/catkin_ws/src/Onboard-SDK-ROS-4.0/src/dji_osdk_ros_obsoleted/include/dji_sdk/")
  --> file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/include/dji_osdk_ros" TYPE DIRECTORY FILES "/home/work/catkin_ws/src/Onboard-SDK-ROS-4.0/include/dji_osdk_ros_obsoleted/dji_sdk/")

line60：
#  file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/share/dji_osdk_ros/launch" TYPE DIRECTORY FILES "/home/work/catkin_ws/src/Onboard-SDK-ROS-4.0/src/dji_osdk_ros_obsoleted/launch")
   --> file(INSTALL DESTINATION "${CMAKE_INSTALL_PREFIX}/share/dji_osdk_ros/launch" TYPE DIRECTORY FILES "/home/work/catkin_ws/src/Onboard-SDK-ROS-4.0/launch")

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2021-07-09 by dji-kyle