dji-sdk/Onboard-SDK-ROS #344: Waypoint TRACE_COORDINATED not working

camilojimo, 2020-06-09:
Hi,

I am using ROS O-SDK with the A3 flight controller, I am trying to upload the wayipoint sample with some variations. I want the drone to continue to continue the trajectory after reaching a waypoint (no stop and continue to the next point) 

in Onboard-SDK-3.9/build/catkin_ws/devel/include/dji_sdk/MissionWaypointTask.h line 307

"uint8 TRACE_COORDINATED = 1       # 1: Coordinated turn mode, smooth transition between waypts, no waypts task\n" is the functionality I want to use 

in my ros node I am integrating the following 

setWaypointDefaults(WayPointSettings* wp)
{
  wp->damping         = 0.2;
  wp->yaw             = 0;
  wp->gimbalPitch     = 0;
  wp->turnMode        = 0;
  wp->hasAction       = 0;
  wp->actionTimeLimit = 100;
  wp->actionNumber    = 0;
  wp->actionRepeat    = 0;
  for (int i = 0; i < 16; ++i)
  {
    wp->commandList[i]      = 0;
    wp->commandParameter[i] = 0;
  }
}

void
setWaypointInitDefaults(dji_sdk::MissionWaypointTask& waypointTask)
{
  waypointTask.velocity_range     = 15;
  waypointTask.idle_velocity      = 15;
  waypointTask.action_on_finish   = dji_sdk::MissionWaypointTask::FINISH_AUTO_LANDING;
  waypointTask.mission_exec_times = 1;
  waypointTask.yaw_mode           = dji_sdk::MissionWaypointTask::YAW_MODE_LOCK;
  waypointTask.trace_mode         = dji_sdk::MissionWaypointTask::TRACE_COORDINATED;
  waypointTask.action_on_rc_lost  = dji_sdk::MissionWaypointTask::ACTION_AUTO;
  waypointTask.gimbal_pitch_mode  = dji_sdk::MissionWaypointTask::GIMBAL_PITCH_FREE;
}

when I try to run this sample I am getting this error:

[ WARN] [1591679266.378069518]: ack.info: set = 3 id = 17
[ WARN] [1591679266.378226679]: ack.data: 231
[ WARN] [1591679266.378349429]: Failed sending waypoint upload command

Can someone please help me with this problem ?

Thanks

lanyusea, 2020-06-09:
231 means the dumping value is incorrect, could you try increasing it maybe to 1

camilojimo, 2020-06-09:
I have try different values from 1 to 10 and I get the same error 
----------------------------------------------------------------------------
[ INFO] [1591682290.565403690]: initialized waypoint mission
[ INFO] [1591682291.578077034]: uploaded the 1th waypoint

[ INFO] [1591682292.586008119]: uploaded the 2th waypoint

[ INFO] [1591682293.598407073]: uploaded the 3th waypoint

[ INFO] [1591682294.605545228]: uploaded the 4th waypoint

[ INFO] [1591682295.617266862]: uploaded the 5th waypoint

[ INFO] [1591682296.623182844]: uploaded the 6th waypoint

STATUS/1 @ getErrorCodeMessage, L656: missionWpUploadCallback
STATUS/1 @ getCMDSetMissionMSG, L883: WAYPOINT_MISSION_CHECK_FAILED
[ INFO] [1591682297.630590267]: uploaded the 7th waypoint

[ INFO] [1591682298.630850592]: waypoint mission initialized and uploaded
-----------------------------------------------------------------------------
[ WARN] [1591682298.631200106]: ack.info: set = 3 id = 17
[ WARN] [1591682298.631291326]: ack.data: 231
[ WARN] [1591682298.631362179]: Failed sending waypoint upload command

is there any other suggestion ???

dji-dev, 2020-06-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35579'>Zendesk ticket #35579</a>:<br><br>This request was closed and merged into request #35460 "Mission sampe sdk 3.9".

matspetter, 2020-08-17:
What does this mean?? Do you have a resolution? Where?

camilojimo, 2020-08-24:
Hi @matspetter I was using ROS 3.8 and this was the solution for my issue on demo_mission.cpp

[image]

closed 2021-07-09 by dji-kyle