dji-sdk/Onboard-SDK-ROS #348: N3 support for 3.8.1

alduxvm, 2020-06-15:
Hi all, 

I know that 4.0 does not support N3 any more (which is actually a horrible decision tbh) but I checkout the 3.8.1 tag and attempted to run the launch file `src/Onboard-SDK-ROS/dji_sdk/launch/sdk.launch` but it appears that even checking out 3.8.1 still the N3 is not supported?? the output is as follows:

```
~/catkin_ws$ roslaunch src/Onboard-SDK-ROS/dji_sdk/launch/sdk.launch
... logging to /home/xxx/.ros/log/ea01efb2-af26-11ea-8c3a-5c5f67f52409/roslaunch-x1-3387.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://x1:40585/

SUMMARY
========

PARAMETERS
 * /dji_sdk/align_time: False
 * /dji_sdk/app_id: xxxx
 * /dji_sdk/app_version: 1
 * /dji_sdk/baud_rate: 921600
 * /dji_sdk/enc_key: xxx...
 * /dji_sdk/serial_name: /dev/ttyUSB1
 * /dji_sdk/use_broadcast: False
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    dji_sdk (dji_sdk/dji_sdk_node)

auto-starting new master
process[master]: started with pid [3397]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ea01efb2-af26-11ea-8c3a-5c5f67f52409
process[rosout-1]: started with pid [3408]
started core service [/rosout]
process[dji_sdk-2]: started with pid [3411]
[ INFO] [1592239226.448320213]: Advanced Sensing is Enabled on M210.

STATUS/1 @ init, L55: Attempting to open device /dev/ttyUSB1 with baudrate 921600...

STATUS/1 @ init, L65: ...Serial started successfully.

STATUS/1 @ parseDroneVersionInfo, L727: Device Serial No. = 0DBDG6F00200J8

STATUS/1 @ parseDroneVersionInfo, L729: Hardware = N3

STATUS/1 @ parseDroneVersionInfo, L730: Firmware = 3.3.8.47

ERRORLOG/1 @ initVirtualRC, L1103: Virtual RC is not supported on this platform!

ERRORLOG/1 @ AdvancedSensing, L58: Please make sure the connected drone is a M210 and firmware version is supported.

[ERROR] [1592239228.663693173]: drone activation error
[ERROR] [1592239228.663815098]: Vehicle initialization failed
```
Is there a way to try to use the onboard SDK for the N3?? if not, why are they still selling a unsupported flight controller?

Thanks!

dji-dev, 2020-06-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/35826'>Zendesk ticket #35826</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for any inconvenience with version update, The N3 supports ROS 3.8.
The following log shows some wrong with advance sensing,  The N3 not support  advance sensing.
so you need to switch off the ADVANCED_SENSING function and modify the code dji_sdk_node.cpp as follows:
 [image]

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

alduxvm, 2020-06-16:
To make it work, I disabled `ADVANCED_SENSING` in https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8/dji_sdk/src/modules/dji_sdk_node.cpp which was enabled by default and after that the activation was successful and I was able to get the topics.

closed 2020-06-16 by alduxvm