dji-sdk/Onboard-SDK-ROS #358: Please help me "cmake fail" , how to fix it?

TopKruZ, 2020-07-13:
I already installed Install DJI SDK and next step Setup the DJI SDK ROS package

but when i "catkin_make" it failed , please help me.

[image]

closed 2020-07-14 by TopKruZ

dji-dev, 2020-07-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36827'>Zendesk ticket #36827</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Have you installed the Dependent Software ? And you can  referenced to : https://developer.dji.com/onboard-sdk/documentation/quickstart/development-environment.html

If the problem remains unsolved, suggest that comfirm and provide the version of OSDK and OSDK-ROS, which should be used with compatible version, such as OSDK4.0 with ROS4.0， OSDK3.9（3.8）with ROS3.8

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

pavankumarbn, 2020-07-14:
@TopKruZ Just run the command: sudo apt-get install ros-"distro"-nmea-msgs

TopKruZ, 2020-07-17:
Hi, I need to ask some question OSDK and OSDK-ROS it support DJI Marvic Air2?

ส่งจาก iPad ของฉัน

dji-dev, 2020-07-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/36827'>Zendesk ticket #36827</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry the OSDK and OSDK-ROS not support DJI Marvic Air2, And the list of OSDK supported models is available on the developer's website:
https://developer.dji.com/products/#!/onboard

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team