dji-sdk/Onboard-SDK-ROS #362: Cant run dji_vehicle_node

lorenzave, 2020-07-21:
I am running the osdk in a tx2 windowless/displayless device. I get prompted an error when trying to launch the new implementation of nodes. I can get the old dji_sdk_node to run normally.
I am running the latest commits for master branch in both OSDK-ROS and OSDK.

`started roslaunch server http://7d4556baec66:36250/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: *************
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 230400
 * /vehicle_node/enc_key: *************
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [5201]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to ac5e3554-cb54-11ea-afc5-0242ac110003
process[rosout-1]: started with pid [5214]
started core service [/rosout]
process[vehicle_node-2]: started with pid [5231]
ERROR: In /build/vtk6-rCT4UR/vtk6-6.2.0+dfsg1/Rendering/OpenGL/vtkXOpenGLRenderWindow.cxx, line 1466
vtkXOpenGLRenderWindow (0x7be0d0): bad X server connection. DISPLAY=[vehicle_node-2] process has died [pid 5231, exit code -6, cmd /usr/src/app/catkin_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/mnt/ssd/logging/ac5e3554-cb54-11ea-afc5-0242ac110003/vehicle_node-2.log].
log file: /mnt/ssd/logging/ac5e3554-cb54-11ea-afc5-0242ac110003/vehicle_node-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
`

dji-dev, 2020-07-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37169'>Zendesk ticket #37169</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

If the sample runs with SSH, you may need to configure the SSH runtime environment for render, as the error shows:

ERROR: In /build/vtk6-rCT4UR/vtk6-6.2.0+dfsg1/Rendering/OpenGL/vtkXOpenGLRenderWindow.cxx, line 1466
vtkXOpenGLRenderWindow (0x7be0d0): bad X server connection. DISPLAY=[vehicle_node-2] process has died [pid 5231, exit code -6, cmd /usr/src/app/catkin_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/mnt/ssd/logging/ac5e3554-cb54-11ea-afc5-0242ac110003/vehicle_node-2.log].

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-09-02:
this has been resolved.

closed 2020-09-02 by lorenzave

dregner, 2022-07-29:
Hello @lorenzave, sorry for long time about it. but I am facing same problem... How do you solved it?