dji-sdk/Onboard-SDK-ROS #367: Waypoint Upload Error

lueninb, 2020-08-05:
hello,

I am having problems running the Waypoint Mission in the included mission_node sample program. We purchased a new **M300** and are running 16.04 on a TX2.

I am doing rosrun dji_osdk_ros mission_node and get the following:

nvidia@tegra-ubuntu:~/catkin_ws$ rosrun dji_osdk_ros mission_node 
| Available commands:                                            |
| [a] Waypoint Mission                                           |
| [b] Hotpoint Mission                                           |
a
[ INFO] [1596665280.014648976]: Creating Waypoints..

[ INFO] [1596665280.016398716]: Waypoint created at (LLA): 48.009042 	-97.931172 	10.000000

[ INFO] [1596665280.017149263]: Uploading Waypoints..

[ INFO] [1596665280.017490135]: Waypoint created at (LLA): 48.009042 	-97.931172 	10.000000
 
[ INFO] [1596665280.018297580]: Waypoint created at (LLA): 48.009071 	-97.931122 	11.000000
 
[ INFO] [1596665280.019149505]: Waypoint created at (LLA): 48.009042 	-97.931073 	12.000000
 
[ INFO] [1596665280.019434984]: Waypoint created at (LLA): 48.009042 	-97.931172 	10.000000
 
[ INFO] [1596665280.020129113]: Initializing Waypoint Mission..

[ WARN] [1596665350.154133753]: ack.info: set = 255 id = 255
[ WARN] [1596665350.154552516]: ack.data: 65535
[ WARN] [1596665350.154781771]: Failed sending waypoint upload command

The output from the rosrun dji_osdk_ros dji_vehicle_node.launch is:

[ INFO] [1596665417.311933981]: initialized waypoint mission
[3232564.021]STATUS/1 @ getErrorCodeMessage, L744: missionWpUploadCallback
[3232564.022]ERRORLOG/1 @ getCommonErrorCodeMessage, L811: NO_RESPONSE_ERROR

[ INFO] [1596665430.838696960]: uploaded the 1th waypoint

[3232577.547]STATUS/1 @ getErrorCodeMessage, L744: missionWpUploadCallback
[3232577.548]ERRORLOG/1 @ getCommonErrorCodeMessage, L811: NO_RESPONSE_ERROR

[ INFO] [1596665444.365141524]: uploaded the 2th waypoint

^C[vehicle_node-2] killing on exit
[3232591.071]STATUS/1 @ getErrorCodeMessage, L744: missionWpUploadCallback
[3232591.072]ERRORLOG/1 @ getCommonErrorCodeMessage, L811: NO_RESPONSE_ERROR

Can someone please help me with this?

dji-dev, 2020-08-06:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37747'>Zendesk ticket #37747</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Have you try to test the sample flight control? or there is only the waypoint mission failed ?
could you please provide the complet terminal log of the rosrun dji_osdk_ros dji_vehicle_node.launch ?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lueninb, 2020-08-06:
Thank you for your help.

Yes, I have tried the other sample codes and they seem to all work EXCEPT **WayPoint Mission**.  Even the Hotpoint Mission works 100%.  Below is the log for **HotPoint which works**.

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1084639
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 921600
 * /vehicle_node/enc_key: 36dfb43c74c3dbc1b...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [6529]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 0d8f5384-d812-11ea-a6b7-00044bf25eeb
process[rosout-1]: started with pid [6542]
started core service [/rosout]
process[vehicle_node-2]: started with pid [6555]
EnableAd: 1
[3305398.909]ERRORLOG/1 @ initVehicle, L165: Failed to initialize ACM Linker channel!
[3305398.919]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH5G00A73M4
[3305398.919]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[3305398.919]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3305398.919]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3305399.919]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[3305400.120]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3305400.120]STATUS/1 @ MopPipelineManagerBase, L48: MOP background task now is created .
[3305401.147]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[3305401.147]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[3305401.148]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[3305401.148]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[3305401.149]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[3305401.149]STATUS/1 @ MopPipelineManagerBase, L51: MOP background task is already created .
[3305401.149]STATUS/1 @ activate, L1329: version 0x304082C

[3305401.160]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1596738269.017612958]: VehicleNode Start
[ INFO] [1596738269.018345540]: Topic startup!
[ INFO] [1596738269.305507743]: Services startup!
[ INFO] [1596738269.516044194]: Use data subscription to get telemetry data!
[ INFO] [1596738269.516368289]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[3305402.740]STATUS/1 @ verify, L244: Verify subscription successful.
[3305402.760]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3305402.760]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[3305402.800]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3305402.800]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[3305402.821]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3305402.821]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[3305402.841]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[3305402.903]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3305402.903]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[... 158 lines cut ...]

[ INFO] [1596737851.784727655]: Uploading Waypoints..

[ INFO] [1596737851.785015336]: Waypoint created at (LLA): 47.873839 	-97.058204 	10.000000
 
[ INFO] [1596737851.785343936]: Waypoint created at (LLA): 47.873868 	-97.058154 	11.000000
 
[ INFO] [1596737851.785615109]: Waypoint created at (LLA): 47.873839 	-97.058104 	12.000000
 
[ INFO] [1596737851.785840564]: Waypoint created at (LLA): 47.873782 	-97.058104 	13.000000
 
[ INFO] [1596737851.786062243]: Waypoint created at (LLA): 47.873753 	-97.058154 	14.000000
 
[ INFO] [1596737851.786278964]: Waypoint created at (LLA): 47.873782 	-97.058204 	15.000000
 
[ INFO] [1596737851.786504643]: Waypoint created at (LLA): 47.873839 	-97.058204 	10.000000
 
[ INFO] [1596737851.786698585]: Initializing Waypoint Mission..

[ WARN] [1596737962.500960327]: ack.info: set = 255 id = 255
[ WARN] [1596737962.501359799]: ack.data: 65535
[ WARN] [1596737962.501603974]: Failed sending waypoint upload command

I appreciate any help you can give me.  I would like to use waypoints with the M300.
thanks

dji-dev, 2020-08-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37747'>Zendesk ticket #37747</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue we have noticed , and submit it to comfirm. if there are any result ,we will contact you soon.
Sorry for any inconvience to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-08-28:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37747'>Zendesk ticket #37747</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the late reply，we have comfirmed that the **WayPoint Mission** (sample:  mission_node, option: a) ,not support the drone M300.
And this note has been added in the code: https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/src/dji_osdk_ros/samples/mission_node.cpp

 [image]
​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/7ogNk1B2wotMa70d8cOxdKSc1' target='zd_attach'>inline1224402337.png</a>

jsampat2, 2020-11-19:
Is the mission waypoint service also unavailable for M300? I am trying to use the mission_waypoint_upload service but it returns False. When would this feature be available for the M300?