dji-sdk/Onboard-SDK-ROS #368: Matrice 210 V2 RTK - Limit yaw velocity and take off speed

manfer33, 2020-08-06:
I am using a Matrice 210 V2 RTK (ROS OSDK 3.8.1) to flight executing a mission with a list of waypoints.  I would like to reduce the velocity params, but I only get modify [this param)(https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8.1/dji_sdk/msg/MissionWaypointTask.msg#L26) to reduce the velocity between waypoints. The yaw velocity (the time when the platform reach a waypoint and heading to the next one) still being too much fast but I don't fin the way to limit it. I would like reduce the takeoff speed too. 

DJI Assistant does not offer any option to config that kind of parameters. Does exist any option throught software? Do I need to modify any file of OSDK?

dji-dev, 2020-08-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37788'>Zendesk ticket #37788</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

There is no direct param or function to set the yaw velocity (the time when the platform reach a waypoint and heading to the next one), but if there are no waypoints task when reaching the waypoint, you can try to set the tracemode to coordinated turn mode, smooth transition between waypoints(https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8.1/dji_sdk/msg/MissionWaypointTask.msg#L16).

and the param damping from the following struct should be set a value great than or equal to 0.2 , if you set the tracemode to coordinated turn mode.
typedef struct WayPointSettings
{

...

  float32_t damping; /*!< Bend length (effective coordinated turn mode only) */
    ...
} WayPointSettings;              // pack(1)

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

manfer33, 2020-08-12:
Thank you for your response. Along the mission our drone must use the Zenmuse onboard. Could be a problem to smooth the yaw velocity when it reaching the waypoint?

By other side, as I ask you in the first message, is there any way to reduce de takeoff speed/acceleration?

dji-dev, 2020-08-13:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37788'>Zendesk ticket #37788</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、Along the mission our drone must use the Zenmuse onboard. Could be a problem to smooth the yaw velocity when it reaching the waypoint?

-->The coordinated turn mode relates to the drone, it is independent of the payload(Zenmuse).

2、 is there any way to reduce de takeoff speed/acceleration?

-->The param velocity_range and idle_velocity can be used to set the Cruising Speed(include the takeoff speed)(https://github.com/dji-sdk/Onboard-SDK-ROS/blob/3.8.1/dji_sdk/msg/MissionWaypointTask.msg#L26  && L27), but there is no param to set the the takeoff speed independent with the Cruising Speed.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team