dji-sdk/Onboard-SDK-ROS #370: M300 Problem with Waypoints

lueninb, 2020-08-07:
hello,

I'm still having problems with waypoints on the **M300**.  I tried Waypoint in sample/mission_node but it does not work (see https://github.com/dji-sdk/Onboard-SDK-ROS/issues/367)

I have also tried the latest example using WaypointV2.  Still it does not work.  I get this error:

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1084639
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 921600
 * /vehicle_node/enc_key: 36dfb43c74c3dbc1b...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [3058]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 20d34122-d8d4-11ea-bfbd-00044bf25eeb
process[rosout-1]: started with pid [3078]
started core service [/rosout]
process[vehicle_node-2]: started with pid [3090]
EnableAd: 1
[3388753.118]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!
[3388753.131]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH5G00A73M4
[3388753.131]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.44
[3388753.131]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3388753.131]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3388754.131]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[3388754.332]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3388754.333]STATUS/1 @ RegisterOSDInfoCallback, L535: register result of geting　FC ground station status pushing : 1

[3388754.333]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[3388754.333]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[3388754.333]STATUS/1 @ activate, L1329: version 0x304082C

[3388754.350]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1596821622.206863262]: VehicleNode Start
[ INFO] [1596821622.207513630]: Topic startup!
[ INFO] [1596821622.542006878]: Services startup!
[ INFO] [1596821622.737650206]: Use data subscription to get telemetry data!
[ INFO] [1596821622.737997822]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[3388755.970]STATUS/1 @ verify, L244: Verify subscription successful.
[3388755.990]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3388755.990]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[3388756.029]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3388756.029]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[3388756.053]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3388756.053]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[3388756.074]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[3388756.134]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3388756.134]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[3388756.151]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3388756.151]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[ INFO] [1596821647.025607197]: called waypointV2GenerateActionsCallback
[ INFO] [1596821647.039729279]: called waypointV2InitSettingCallback
[3388780.225]ERRORLOG/1 @ initWaypointV2, L1946: Init mission setting ErrorCode:0xFFFFFFFF

[3388780.226]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[3388780.226]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[3388780.226]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.

Please help me.  I really need to use waypoints with the M300.
thanks

dji-dev, 2020-08-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37826'>Zendesk ticket #37826</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry for the inconvenience to you, the https://github.com/dji-sdk/Onboard-SDK-ROS/issues/367 issue has reproduced and submit to further comfirm. But the waypoint v2 demo works normally.  Suggest that you can try again with OSDK master and OSDK ROS master again.

Notice: The Master branch is constantly being updated, and suggest test in the simulator.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lueninb, 2020-08-10:
hello,

Yes, I have updated BOTH the OSDK and the OSDK ROS using master to the latest commits and it does not work.  I still get the same error message as shown above.

osdk is updated to commit a4dae2412aa055ab9d2b490cde0d441bca299ba8.
osdk ROS is updated to commit 9a556f08efb6318a96f79ec499f5c05a20143958

Please advise me what other steps to try.
Thanks again for your help.

dji-dev, 2020-08-11:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/37826'>Zendesk ticket #37826</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to the logs,
[3388753.118]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!
It seems that your Dual-A USB cables unconnected or works abnormal. Suggest that you check the cables and then test the waypoint v2

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

jsampat2, 2020-11-23:
Hello,
I tried running the WaypointV2 sample as well, but I ran into the following error, is there a certain way to run the sample that I need to know of?
[image]

Thank you!
Have a nice day

AkshatAgarw, 2022-01-21:
Hi there,
i am having the same issue and same log file while executing the waypoint mission sample.
I want to run the waypoint mission but without advance sensing feature can you please suggest me what should i do for that.

Thanks