dji-sdk/Onboard-SDK-ROS #374: not able to start up the  dji_vehicle_node

ynaka01-max, 2020-08-17:
Now I am using Matrice210 v2 and manifold2-G.
When  I execute command as below.
```
$ rosed dji_osdk_ros dji_vehicle_node.launch
```

I got 
```
started roslaunch server http://manifold2:43289/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vehicle_node/acm_name: /dev/ttyTHS2
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: **********
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 230400
 * /vehicle_node/enc_key: *******************...
 * /vehicle_node/serial_name: /dev/ttyTHS2
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [10286]
............................
[4256287.781]STATUS/1 @ adv_pthread, L46: [4256287.781]STATUS/1 @ adv_pthread, L48: adv pthread running !!!!!!!!!!!!!!!!!!!!!!!
[4256287.792]STATUS/1 @ activate, L1353: Activation successful

[vehicle_node-2] process has died [pid 10316, exit code -11, cmd /home/dji/catkin_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/home/dji/.ros/log/625a0a40-e0b8-11ea-83fc-61501fff8c85/vehicle_node-2.log].
log file: /home/dji/.ros/log/625a0a40-e0b8-11ea-83fc-61501fff8c85/vehicle_node-2*.log
```
and log 625a0a40-e0b8-11ea-83fc-61501fff8c85 is as below.

```
............................
[roslaunch.parent][INFO] 2020-08-17 18:01:13,912: ... roslaunch parent running, waiting for process exit
[roslaunch][INFO] 2020-08-17 18:01:13,913: spin
[roslaunch][ERROR] 2020-08-17 18:01:16,392: [vehicle_node-2] process has died [pid 9370, exit code -11, cmd /home/dji/catkin_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/home/dji/.ros/log/625a0a40-e0b8-11ea-83fc-61501fff8c85/vehicle_node-2.log].
log file: /home/dji/.ros/log/625a0a40-e0b8-11ea-83fc-61501fff8c85/vehicle_node-2*.log
[roslaunch.pmon][INFO] 2020-08-17 18:01:16,395: ProcessMonitor.unregister[vehicle_node-2] starting
[roslaunch.pmon][INFO] 2020-08-17 18:01:16,396: ProcessMonitor.unregister[vehicle_node-2] complete

```

Log told process has died.
Please teach me how to solve it.

dji-dev, 2020-08-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38110'>Zendesk ticket #38110</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

May I comfirmed with you that the app_id and app_key is correctly applied from the DJI developer website?

and then the acm_name should be corrected , such as /dev/ttyACM0, depends on your acm device name (with dual A USB cable connected)
​  [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/J02XKaZoWeLIMrGc3tqlgJzC9' target='zd_attach'>inline1656701454.png</a>

ynaka01-max, 2020-08-19:
Thank you. The problem was solved.

closed 2020-08-19 by ynaka01-max