dji-sdk/Onboard-SDK-ROS #377: Turning off display for headless systems

wiktorpi, 2020-08-19:
Hi,
I have an issue with DJI OSDK ROS 4.0. When trying to launch dji_vehicle_node.launch i get a VTK error 

> ERROR: In /build/vtk6-5rzKol/vtk6-6.3.0+dfsg1/Rendering/OpenGL/vtkXOpenGLRenderWindow.cxx, line 1475
vtkXOpenGLRenderWindow (0x559e48a300): bad X server connection. DISPLAY=[vehicle_node-2].

The system is headless and therefore cannot provide a window. Looking in the code however, i could not find a way to disable this unnecessary display. I have tracked it down to the PointCloudViewer.
```
#5  0x0000007fb7db4460 in M210_STEREO::PointCloudViewer::PointCloudViewer(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) () at /home/ubuntu/cbuild_ws/devel/lib/libdji_osdk_ros.so
```
Is there a way to disable it that I am missing?

Wiktor

dji-dev, 2020-08-20:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38194'>Zendesk ticket #38194</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue has fixed and updated in master branch, if you have not installed the opencv related viz lib , it can be work with ssh to launch dji_vehicle_node.launch. and if you have installed the relative lib , and then you should use: ssh -X username@XXXX to launch dji_vehicle_node.launch

It is recommanded that you update the OSDK and ROS again from the master branch.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

wiktorpi, 2020-09-16:
Issue is in fact solved after pulling the recent commits. Thank you!

closed 2020-09-16 by wiktorpi

eduardofamendes, 2022-09-09:
I am having this same problem today (Sep 9, 2022), with dji-osdk-ros 4.1.0. Any update on that? I can run it by ssh with "-X" option, but I cannot set it to auto-start as a ubuntu service, because it does not work without the X.