dji-sdk/Onboard-SDK-ROS #379: How to run sample in real world flight.

ynaka01-max, 2020-08-19:
I started up flight control sample according to the procedure.
I also confirmed that the drone moves according to the code on the simulator correctly.
Therefore, I want to fly the drone outside and confirm it works correctly.
However, I do not know how to run service remotely.
Could you give me some information on that?

Drone: Matrice 210 V2
Onboard PC: Manifold 2-G
Ground PC: Mac

pavankumarbn, 2020-08-20:
Hi ynaka01-max, You can use either putty or use: ssh -X username@XXXX to run service remotely in outside. I have used putty as shown in this video ( https://www.youtube.com/watch?v=2WMTu2o1iq4)

ynaka01-max, 2020-08-20:
Thank you. 
The problem is resolved.

closed 2020-08-20 by ynaka01-max