dji-sdk/Onboard-SDK-ROS #382: How to supply power from drone to manifold

ynaka01-max, 2020-08-24:
I don't know how to supply power from drone to manifold.
I connect drone and manifold with XT30U-F cable.
However, no power supplied.
Would you please tell me?

Drone: Matrice 210 v2
Manifold: Manifold2-G

lorenzave, 2020-08-24:
you need to activate it the power output of the IO through the remote. Drone options, bottom I/O Options I believe.

pavankumarbn, 2020-08-24:
@ynaka01-max They will provide (within the box) a cable to draw power from Drone battery.  haven't you received it? I received it for DJI Matrice 100 and Manifold.

closed 2020-08-25 by ynaka01-max

dji-dev, 2020-08-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38355'>Zendesk ticket #38355</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

You can check the settings of RC and enable the Power Supply port.
 [image]​
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/ACDi3dbiQgQOSTkptWnxCNsle' target='zd_attach'>inline-144980286.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/F92Ctdh4to5QgmYhZjXYHQIvj' target='zd_attach'>inline-1940569089.png</a>