dji-sdk/Onboard-SDK-ROS #388: ROS node dies when starting camera stream

lorenzave, 2020-09-01:
Hi,
I am using M300 with latest firmware and 4.0.1 OSDK and OSDK-ROS. I am using the obsolete interface.
Whenever I call the setup_camera_stream service node dies and I cannot get any images.

Thanks

dji-dev, 2020-09-02:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38717'>Zendesk ticket #38717</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Is it test with OSDK-ROS sample ？ if runs with sample,  could you please provide the detail operation steps and the complete terminal log.
if not, suggest you provide the detail interface you used and test with origin sample.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-09-02:
Hi Support,

This is with the "obsolete" interface which doesn't have any samples. I made an error while trying to launch the node, now **I CAN** setup the camera and get frames in ROS. 

The issue now is when I call `rosservice call /dji_osdk_ros/setup_camera_stream "cameraType: 1 start: 0"` to stop the stream the node dies. I can mitigate it by forcing a respawn to the node but it would be nice to solve it.

Thanks

dji-dev, 2020-09-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38717'>Zendesk ticket #38717</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

For better technical support and maintenance, it is recommended that you do not use obsolete interfaces.
Sorry for the inconvience to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzmhe, 2020-09-07:
A colleague of @lorenzave here.

We use the obsolete interface because neither 4.0.0 or 4.0.1 have support for velocity control and does not publish information on vital topics as detailed in #389 .

dji-dev, 2020-09-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38717'>Zendesk ticket #38717</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The obsolete interface is also available now, for match the older versions(3.8) of ROS Node. And the version 4.0.0 or 4.0.1 supports velocity control also, but it maybe not provide the complete demo in vehicle node.You can refer to the obsolete interface for self-implementation.

If you need the DJI provide relevant demo, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team