dji-sdk/Onboard-SDK-ROS #390: Vehicle node unstable connection

lorenzave, 2020-09-07:
Hi Support,

whenever I'm developing the vehicle_node (new interface) looses randomly connection to the device.
Sometimes it takes a couple of minutes, sometimes 20/30m, and sometimes is able to be connected during hours. 
Do you have any idea what may be happening?

```
[1758614.211]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1599486449.344254161]: VehicleNode Start
[ INFO] [1599486449.344433171]: Topic startup!
[ INFO] [1599486449.451786714]: Services startup!
[ INFO] [1599486449.519687333]: Use data subscription to get telemetry data!
[ INFO] [1599486449.519774822]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[1758615.451]STATUS/1 @ verify, L244: Verify subscription successful.
[1758615.471]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[1758615.471]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[1758615.494]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[1758615.494]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[1758615.522]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[1758615.522]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[1758615.545]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[1758615.601]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[1758615.601]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[1758615.643]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[1758615.643]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[1758764.478]STATUS/1 @ getCameraList, L199: register result of geting camera pushing : 1

[1758764.506]STATUS/1 @ getCameraList, L218: Request start pushing camera info ack = 0

[1758765.527]STATUS/1 @ getCameraList, L235: Request end pushing camera info ack = 0

[1758765.527]STATUS/1 @ getCameraList, L236: mounted	cam0:1 cam2:0 cam3:0
[1758765.527]STATUS/1 @ getCameraList, L237: type   	cam0:43 cam2:0 cam3:0
[1758765.527]STATUS/1 @ startH264Stream, L457: camera[0] is mounted

subsrcibe data success!
[1758874.433]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758875.433]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758876.434]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758877.435]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758878.438]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758879.439]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
[1758880.440]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!

```

Thanks.

dji-dev, 2020-09-08:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This issue needs further confirmation , could you please provide the relevant info:
1，The baudrate used for ROS
2，Drone Type and firmware version
3、operate steps to reproduce it

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-09-08:
> Agent comment from kyle.cai in [Zendesk ticket #38909](https://djisdksupport.zendesk.com/agent/tickets/38909):
> 
> Dear developer ,
> 
> Thank you for contacting DJI.
> 
> This issue needs further confirmation , could you please provide the relevant info:
> 1，The baudrate used for ROS

230400. I am using the same hardware setup we have been using for the M210 for over 2+ years with no issues.

> 2，Drone Type and firmware version

M300 FW: 01.00.0211
DJI OSDK: 4.0.1
DJI OSKD-ROS: 4.0.1

> 3、operate steps to reproduce it

This error is hard to reproduce reliably as sometimes it happens without doing anything just starting the node and letting it sit. Sometimes Im able to start h264 stream for a long time 1h+. Usually when it happens is after 3/4 minutes of being suscribed to the h264 stream and Im not able to recover the connection. I need to restart the node. When I do this sometimes the 50Hz package is not able to start.

```
[1821861.017]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1599549696.160032613]: VehicleNode Start
[ INFO] [1599549696.160321031]: Topic startup!
[ INFO] [1599549696.266970570]: Services startup!
[ INFO] [1599549696.350405540]: Use data subscription to get telemetry data!
[ INFO] [1599549696.350677510]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[1821862.264]STATUS/1 @ verify, L244: Verify subscription successful.
[1821862.283]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[1821862.283]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[1821862.306]STATUS/1 @ startPackage, L345: Start package 2 result: 4.
[1821862.306]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[1821862.306]STATUS/1 @ getErrorCodeMessage, L744: startPackage
[1821862.306]STATUS/1 @ getCMDSetSubscribeMSG, L828: SUBSCRIBER_PACKAGE_ALREADY_EXISTS

[1821862.328]STATUS/1 @ removePackage, L457: Remove package 2 successful.
[ERROR] [1599549696.440868082]: Failed to start 50Hz package

```

Hope this feedback helps fixing the bug.

Thanks.
> 
> Thank you for your understanding and support, hope you have a nice day.
> 
> Best Regards,
> DJI SDK Support Team

rebellojason7, 2020-09-25:
I have the same issue. Any idea what the solution might be ?

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About the error ：
[1758874.433]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!

This is the communication heartbeat between OSDK and the drone is disconnected. and it related the serial communication. If this problem occurs frequently, it is recommended to increase the baud rate and check whether the application logic causes the heartbeat to be disconnected.

and  If it is convenient, you can provide the following information and confirm it together.
1，The baudrate used for ROS
2，Drone Type and firmware version
3、operate steps to reproduce it

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzave, 2020-09-25:
@dji-dev I responded two weeks ago with that info with no response...
I have been able to reproduce it with the M300, did you change the way the heartbeat works? With the m210v2 and obsolete we never had similar issues, only have seen it with new model and new interface, using the same hardware as before. Makes me believe its more related to the OSDK or the wrapper.

Best regards.

dji-kyle, 2020-09-25:
> @dji-dev I responded two weeks ago with that info with no response...
> I have been able to reproduce it with the M300, did you change the way the heartbeat works? With the m210v2 and obsolete we [...]

Sorry, due to some reasons for the mail import, we have ignored your feedback, and we will reconfirm the information.

rebellojason7, 2020-09-25:
@dji-dev @dji-kyle 
1，The baudrate used for ROS - 921000
2，Drone Type and firmware version - Matrice 210 V2 with RTK (recently purchased) , 4.0.1
3、operate steps to reproduce it - The moment I run the OSDK-ROS node, after 3-4 minutes I get the error. Nothing special done during operation.
How does one go about solving it ?

dji-kyle, 2020-09-27:
@lorenzave Sorry, notcied that the baudrate : 230400 you used, for ROS , the buad rate: 921600 or above is recommanded,And because the OSDK4.0 heartbeat mechanism is added, it is necessary to ensure sufficient serial communication bandwidth.

In addition, according to your feedback, the problem is difficult to reproduce. It is recommended that you increase the baud rate first to see if the recurrence probability decreases.

dji-kyle, 2020-09-27:
> @dji-dev @dji-kyle
> 1，The baudrate used for ROS - 921000
> 2，Drone Type and firmware version - Matrice 210 V2 with RTK (recently purchased) , 4.0.1
> 3、operate steps to reproduce it - The moment I run [...]

@rebellojason7  Could you please check the firmware version with DJI Assitant2 and OSDK(non-ROS) version ?
I test this issue with 
1、M210 RTK v2(firmware: v01.00.710) , 
2、baudrate：921600 
3、OSDK4.0.1  +   OSDK-ROS 4.0.1
4、roslaunch dji_osdk_ros dji_vehicle_node.launch and rosservice call setup_camera_stream 0 1
failed to reproduce it.

Is the operation different from yours?

rebellojason7, 2020-09-29:
We have the Manifold 2G using DJI Onboard SDK and ROS SDK v4.0.1. The drone firmware is at v01.00.710. We run 3 camera services:
`rosservice call setup_camera_stream 1 1` to connect to Zenmuse X5S
`rosservice call setup_camera_stream 0 1` for FPV
`rosservice call stereo_vga_subscription 0 1 0` for stereo VGA stream
Along with running the camera streams, we are publishing gimbal commands using the GimbalAction service.

To connect to the flight controller, we use a USB to serial cable with the PT2303 chip. We saw a forum post that says to use an FT232R chip but the same post says it is not supported on Manifold 2G (https://forum.dji.com/forum.php?mod=viewthread&tid=213098). We can connect to the drone at 921600 baud rate and confirmed the baud rate is correct using DJI Assistant 2.

We get the error below after running for a period of time (5-15 mins). This doesn't always happen so it is hard to reproduce on our drone.
```
Error:
[swscaler @ 0x7f1c140c00] No accelerated colorspace conversion found from yuv420p to rgb24.
STATUS/1STATUS/1STATUS/1STATUS/1STATUS/1[3162107.589]ERRORLOG/1 @ storeData, L193: buffer overflow
Decoder Callback Thread: Get image time out

OSDK lost connection with Drone!
[3162168.437]STATUS/1 @ fcLostConnectCallBack, L1502: [3162170.057]STATUS/1 @ fcLostConnectCallBack, L1502: [3162171.677]STATUS/1 @ fcLostConnectCallBack, L1502: [3162173.301]STATUS/1 @ fcLostConnectCallBack, L1502: OSDK lost connection with Drone!
Decoder Callback Thread: Get image time out
```

dji-dev, 2020-09-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because the heartbeat of OSDK is related to serial communication, if the problem is not always happen, it is recommended to check the serial communication problem.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

rebellojason7, 2020-10-01:
@dji-dev @dji-kyle  We realise its a serial communication problem. The question is how do we go about debugging what is causing it, not what the problem is about. The drone is not usable since the serial communication problem causes the attitude and gimbal angles to stop publishing. We will be forced to send the drone back and get a refund if this problem cannot be solved.

@lorenzave did you manage to solve the problem ? any help would be appreciated

mhoejgaard, 2020-10-01:
@rebellojason7 We'll get a response to you on the email your colleague sent us tomorrow danish time.

rebellojason7, 2020-10-01:
Thanks @mhoejgaard 

@dji-kyle @dji-dev please let us know if you also have any solutions

lorenzave, 2020-10-01:
Hey,

@dji-dev @dji-kyle 
We tried using a higher baud (921600) and it seems to do the job. I had the dji_vehicle_node running for 4/5h no problem. 

@rebellojason7 
Your issue seems similar to ours. There is a heartbeat signal sent to the drone and I'd recommend measuring TX (for the heartbeat) and RX (for all the telemetry). The bandwidth is really needed as it pushed to the limits at that baud so if you want to run all the topics its a must. You can also remove some of the topics to relief the line. It should be clearly stated that it only works for 921600+. The drone/DJI firmware seems to also be pretty picky with the heartbeat signal with a low timeout. Once it disconnects it's not able to recover and you need to reset the node. So grab an oscilloscope and look for clean square waves :D 

Maybe DJI should make the timeout configurable/a bit longer and maybe re-establish the communication once it's lost.

dji-kyle, 2020-10-03:
> Thanks @mhoejgaard
> 
> @dji-kyle @dji-dev please let us know if you also have any solutions

@rebellojason7 if you use the manifold 2G you can try to use baud rate: 1000000 and test it.
[image]

rebellojason7, 2020-10-06:
@dji-dev @dji-kyle we put the baud rate to 1000000 and still experience the issue. Is there a way we can get on a call with someone from dji and sort this issue. We have reached our limit with regard to trying to solve this issue and will need to send it back for a full refund if we cant solve it. Please let us know when can we get on a call and solve this issue. THanks

dji-dev, 2020-10-07:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

@rebellojason7
Sorry, I want to confirm with you again, is there an exception only when using the GimbalAction service? If only run the camera streams without using GimbalAction service, will there be an exception? In addition, whether the steps for triggering the GimbalAction service are also easy to provide, I will try to reproduce it again.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

rebellojason7, 2020-10-08:
Hello @dji-dev @dji-kyle 
It is quite possible that the exception is thrown when running the GimbalAction service, I have not tested whether I get an exception after running only the cameras. I recently got this error. Please see attached image 
[image]
Some steps to reproduce the error
1. I run the launch file given in the dji ros osdk launch folder.
2. I call the service to start the main camera stream
3. I then call my own code which basically reads a bunch of encoder angles from a text file and calls the GimbalAction service to move the gimbal. This code was built off the sample provided by dji in the gimbal_camera_control_node.cpp case 'm'.
4. I cannot guarantee a time by which it will start to show the error as it doesnt seem to happen at the same time. Please let me know if you would like me to provide you with my code that I use and let me know what you find.
THanks,
Jason.

dji-dev, 2020-10-09:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to the information you provided, I used the following steps to test for nearly an hour, but the problem still could not be reproduced. It is recommended that you use this method to verify whether the same exception occurs.

1、M210 RTK v2 (firmware: v01.00.0710) + OSDK 4.0.1 + ROS 4.0.1 + x5s (01.07.00.45）
2、roslaunch dji_osdk_ros dji_vehicle_node.launch
3、rosservice call setup_camera_stream 1 1
4、rosrun dji_osdk_ros gimbal_camera_control_node

and then alternate m and n options to run GimbalActions，but within an hour of the test, there was no abnormal heartbeat disconnection

test result:
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/VPBl22A6iEQSaR3gpxuT2wodS' target='zd_attach'>inline-1409008554.png</a>

rebellojason7, 2020-10-09:
@dji-dev @dji-kyle 
So I ran the same test as you on my configuration which is listed below and found these results
1. M210 RTK V2 (firmware: v01.00.0710) + OSDK 4.0.1 + ROS 4.0.1 + x5s (01.07.00.45), exact same firmware
2. OSDK-ROS last commit (Merge pull request #383 from DJI-Colin/update_readme_about_4.0.1_0824)
3. OSDK last commit (Merge pull request #619 from dji-sdk/master. OSDK 4.0.1 release on OSDK 4.0 branch)
4. roslaunch dji_osdk_ros dji_vehicle_node.launch
5. rosservice call setup_camera_stream 1 1
6. rosrun dji_osdk_ros gimbal_camera_control_node
7. I have a manifold 2G connected to wallpower.

After hitting m and n and seeing the gimbal move I got this output below 
[image]

After running for 3 minutes I saw this output with 
Decoder Callback thread: Get image time out
[image]

Then I saw the fcLostConnectionCallback as seen below
[image]

Please let us know when we can get on a call and sort the issue as waiting 2 days with a reply isnt getting us far with each update.

dji-dev, 2020-10-10:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/38909'>Zendesk ticket #38909</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, we are currently unable to provide telephone technical support. Regarding this problem, the results of the assistance investigation are as follows:
1. According to the screenshot you provided at the end, please confirm whether the operation steps you provided are missing? （As the following picture）
    The picture shows that using "rostopic echo -c *" to get the TOPIC data.
     [image]​

2、The TOPIC data subscribtion is through serial communication, which will occupy the bandwidth of the serial port. If you make changes to this part, it may affect the heartbeat data.

3、about the error：
    Decoder Callback thread: Get image time out
    The video stream transmitted via USB cable, If the gimbal and camera are operated while acquiring the video stream, it will occured probabilistically but it should not be the cause of the abnormal heartbeat. (Currently it can be reproduced to "Get image time out", but it did not cause abnormal heartbeat disconnection).

4、At present, the original sample of OSDK (ROS) cannot be used to reproduce your problem. Is the original sample verified under the test operation steps you provide?

And then I also using the orgin sample test it with
1、 roslaunch dji_osdk_ros dji_vehicle_node.launch
2、rosservice call setup_camera_stream 1 1
3、rostopic echo -c dji_osdk_ros/imu
4、rostopic echo -c dji_osdk_ros/gimbal_angle
5、rosrun dji_osdk_ros gimbal_camera_control_node

 [image]​

suggestions：
1、In order to better distinguish and troubleshoot problems, it is recommended that you directly use **orgin Sample** first and use the above steps to verify whether the problem still occurs，and then check for problems in use

2、Replace the serial port connection module to test, check whether there is serial communication data loss.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/LpIaVD5j2rKdgkd9qLTtHMx0d' target='zd_attach'>inline905207543.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/mJlua5U5jia7lqmNW98TA5ugr' target='zd_attach'>inline2114189517.png</a>

tuyaliang, 2020-11-13:
I have the same error
1、M300 RTK (firmware: 3.4.8.56) ,
2、baudrate：230400
3、OSDK4.0.1 + OSDK-ROS 4.0.1

Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.1.63:42677/

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.9
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1093670
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 230400
 * /vehicle_node/enc_key: 7a4c81ac71d2e669e...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [23647]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a0eb3768-2558-11eb-ae3d-00044ba56dc2
process[rosout-1]: started with pid [23658]
started core service [/rosout]
process[vehicle_node-2]: started with pid [23665]
EnableAd: 1
[3212017.652]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3212017.652]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3212017.652]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[3212017.654]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[3212018.654]STATUS/1 @ sendHeartbeatToFCTask, L1576: OSDK send heart beat to fc task created.
[3212018.854]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[3212018.856]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[3212018.857]STATUS/1 @ Firewall, L46: Firewall is initializing ...
[3212018.871]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1ZNDH9300BGBU3
[3212018.871]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.8.56
[3212018.897]STATUS/1 @ AdvancedSensing, L110: Start bulk channel for M300's liveview!
[3212018.897]STATUS/1 @ AdvancedSensing, L117: Create task for M300's liveview!
[3212018.900]STATUS/1 @ AdvancedSensing, L125: Start bulk channel for M300's perception
[3212018.900]STATUS/1 @ AdvancedSensing, L132: Create task for M300's advanced sensing!
[3212018.900]STATUS/1 @ AdvancedSensing, L135: Advanced Sensing init for the M300 drone
[3212018.900]STATUS/1 @ LiveViewImpl, L89: Finding if liveview stream is available now.
[3212019.701]STATUS/1 @ init, L254: Start advanced sensing initalization
[3212019.701]STATUS/1 @ activate, L1309: osdk policy file updating(1) ......
[3212020.702]STATUS/1 @ activate, L1317: osdk policy file updating(2) ......
[3212020.708]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:0
[3212020.708]STATUS/1 @ RequestUploadPolicyFileHandle, L163: Upload policy file info md5 checksum and version
[3212020.710]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3212020.710]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 0 0 200
[3212020.711]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:1
[3212020.711]STATUS/1 @ RequestUploadPolicyFileHandle, L183: request upload policy file data: 1 200 169
[3212020.714]STATUS/1 @ RequestUploadPolicyFileHandle, L146: request upload policy file type:2
[3212020.715]STATUS/1 @ RequestUploadPolicyFileHandle, L211: request upload policy file success
[3212021.703]STATUS/1 @ activate, L1329: version 0x3040838

[3212021.728]STATUS/1 @ activate, L1367: Activation successful

[ INFO] [1605234824.161162774]: VehicleNode Start
[ INFO] [1605234824.171561439]: Topic startup!
[ INFO] [1605234824.241248096]: Services startup!
[ INFO] [1605234824.270057631]: Use data subscription to get telemetry data!
[ INFO] [1605234824.270133919]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[3212022.888]STATUS/1 @ verify, L244: Verify subscription successful.
[3212022.909]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[3212022.909]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[3212022.931]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[3212022.931]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[3212022.963]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3212022.963]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[3212022.980]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[3212023.019]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[3212023.019]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[3212023.040]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[3212023.040]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[3212033.318]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212034.942]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212036.565]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212038.181]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212039.795]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212041.421]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!
[3212043.037]STATUS/1 @ fcLostConnectCallBack, L1532: OSDK lost connection with Drone!

closed 2021-02-25 by lorenzave