dji-sdk/Onboard-SDK-ROS #393: Failed to initialize ros dji_vehicle_node.launch in M300

yuuud1102, 2020-09-12:
Hi,
I installed the Onboard SDK 4.0.1 and I am getting an error when I launch 
ros dji_vehicle_node.launch

```
dji@manifold2:~$ roslaunch dji_osdk_ros dji_vehicle_node.launch 
... logging to /home/dji/.ros/log/175324ae-f4f0-11ea-ac5d-60601fcd5d40/roslaunch-manifold2-3396.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://manifold2:42677/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1093896
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 1000000
 * /vehicle_node/enc_key: edc7c0e822109e821...
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [3406]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 175324ae-f4f0-11ea-ac5d-60601fcd5d40
process[rosout-1]: started with pid [3419]
started core service [/rosout]
process[vehicle_node-2]: started with pid [3423]
EnableAd: 1
[2184429.932]ERRORLOG/1 @ initVehicle, L154: Failed to initialize Linker channel
Init Vehicle Failed, TERMINATE!
[vehicle_node-2] process has died [pid, exit code 1, cmd /home/dji/catkin_ws/devel/lib/dji_osdk_ros/dji_vehicle_node __name:=vehicle_node __log:=/home/dji/.ros/log/175324ae-f4f0-11ea-ac5d-60601fcd5d40/vehicle_node-2.log].
log file: /home/dji/.ros/log/175324ae-f4f0-11ea-ac5d-60601fcd5d40/vehicle_node-2*.log

```

I have already enabled API control DJI Assistant 2 for Matrice,
and tried different pare of baud rate, all of them showed same result.
BTW  I can`t found log file named "/home/dji/.ros/log/175324ae-f4f0-11ea-ac5d-60601fcd5d40/vehicle_node-2*.log" in that error message.

Here's my other status:
osdk version; 4.0.1
onbord computer; manifold 2-G
ubuntu:16.04
Drone; matrice 300 rtk

Any help please!

dji-dev, 2020-09-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39094'>Zendesk ticket #39094</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Suggest check the device connections, as the log shows, does it rassociated with the USB port (USB-TTL) ?
 [image]​

And there is a link about device connection, maybe give you some advices.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/25SGBBSYD76pU4iddKSuFtVWc' target='zd_attach'>inline-1359739166.png</a>

yuuud1102, 2020-09-14:
Thank you for your reply.

Can you post the link again, please?

dji-dev, 2020-09-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39094'>Zendesk ticket #39094</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

So sorry for missing the link,
https://developer.dji.com/document/1bb1b2dc-9694-4d6b-ad43-fa69c5a10400

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

yuuud1102, 2020-09-14:
I used the a USB to TTL cable and it worked!

Thank you!!

GH1413, 2022-04-18:
Hi, i have the same problem as the one stated above. Regarding the device connections, how do i check that they are associated with the USB port?