dji-sdk/Onboard-SDK-ROS #394: Sample flight control

ynaka01-max, 2020-09-14:
Sample code flight_control_node.cpp's moveByPosOffset move drone according to Ground (World) Coordinate System(NEU), however, I want to move drone according to body coordinate system (DRU). How should I change the code below？
```
bool moveByPosOffset(FlightTaskControl& task, MoveOffset&& move_offset)
{
  task.request.task = FlightTaskControl::Request::TASK_GO_LOCAL_POS;
  // pos_offset: A vector contains that position_x_offset, position_y_offset, position_z_offset in order
  task.request.pos_offset.clear();
  task.request.pos_offset.push_back(move_offset.x);
  task.request.pos_offset.push_back(move_offset.y);
  task.request.pos_offset.push_back(move_offset.z);
  // yaw_params: A vector contains that yaw_desired, position_threshold(Meter), yaw_threshold(Degree)
  task.request.yaw_params.clear();
  task.request.yaw_params.push_back(move_offset.yaw);
  task.request.yaw_params.push_back(move_offset.pos_threshold);
  task.request.yaw_params.push_back(move_offset.yaw_threshold);
  task_control_client.call(task);
  return task.response.result;
}
```

dji-dev, 2020-09-15:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39151'>Zendesk ticket #39151</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

You can't set the Coordinate System directly by the demo: flight_control_node.cpp.  The flight control API: flightCtrl （OSDK API））, which used to change the Coordinate System.

About the ROS invoke OSDK API，you can refer to the vehicle_wrapper.cpp:
bool VehicleWrapper::moveByPositionOffset(ACK::ErrorCode& ack, int timeout, MoveOffset& p_offset)
{

...

//OSDK API

vehicle->control->positionAndYawCtrl(xCmd, yCmd, zCmd, yawDesiredRad / DEG2RAD);

...

}

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

lorenzmhe, 2020-11-24:
You can also use the flightCtrl method in vehicle->control. It gives you the flexibility to use whichever mode you desire.