dji-sdk/Onboard-SDK-ROS #395: about flight control values by flight_control_setpoint_ENUposition_yaw

norihiro-ito, 2020-09-17:
Hi, I'm trying to control a A3 based drone by this SDK.
I'm developing codes with the DJI simulator now.
After I got  the authority of FC, I can flight the drone by  flight_control_setpoint_ENUposition_yaw topic publishing.
But I don't know the measurement unit of control values.
Joy axes [5.0, 0.0, 0.0, 0.0, 0.0] means 'Go 5.0 meter to heading'. Is this OK?
So I monitored local_position topic using set_local_pos_ref() service.
I guess set_local_pos_ref() service reinitialize local coordinate system to [0, 0, 0].  wrong?
But the result of axes [5.0, 0.0, 0.0, 0.0, 0.0] caused x: about 2.5,,, moving to drone heading.
local_position's measurement unit is 'meter'? also axes?
If both is same unit, why unmatch the values?
I guess the reason maybe indoor GPS variances cause it. But alway? or both are different units?

And I'd like to know when in outdoor and with RTK module, will the results improve?

dji-dev, 2020-09-17:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39246'>Zendesk ticket #39246</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The measurement unit is 'meter'. and GPS signal has error range, and there may be bad INDOOR GPS signal. However, according to the information you provided, it cannot be determined temporarily caused by the GPS signal . It is suggested that you test it outdoor and comfirm whether the situation is affected by GPS variances or not.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

norihiro-ito, 2020-09-18:
DJI team,Thank you for your rapid response.
I got the measurement units as meter. 
Isn't the API usage I made wrong?
At the moment, I have determined that caused by indoor GPS variances.
I will test this more outdoor.

And I also wish the information of the RTK variances order.
Best regards,
nito

dji-dev, 2020-09-18:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39246'>Zendesk ticket #39246</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Suggest debug the code in detail and verify this issue，if test it in the simulator，
There should be little relationship with GPS signal or error.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

norihiro-ito, 2020-09-23:
DJI team, thanks your advices. I will do it.

But I feel your words not clearly answer to my question.
Does it means the simulator cannot simulate GPS signal enough?
My DJI Go app shows a warning message on the tablet screen.
It is like 'As GPS signal is too weak, cannot take correct position.'(actually in Japanese).

Please teach me yes/no to following.
(1) Does set_local_pos_ref() service reinitialize local coordinate system to [0, 0, 0, 0] ?
(2) Should a Joy axes [5.0, 0.0, 0.0, 0.0] cause  local_posion x: 5.0 ? 

nito

dji-dev, 2020-09-23:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39246'>Zendesk ticket #39246</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I am sorry the description may not be clear enough，
1、Does it means the simulator cannot simulate GPS signal enough?
    --> Yes, When starting the simulation, the actual GPS signal will not be used, that is, the actual GPS signal does not affect the simulated flight test.  and the remote control should no longer report a weak GPS signal, when starting the simulation.

2、Does set_local_pos_ref() service reinitialize local coordinate system to [0, 0, 0, 0] ?
   -->Not exactly correct，according to the  code , set the local postion to current_gps_*
 [image]​

3、 Should a Joy axes [5.0, 0.0, 0.0, 0.0] cause local_posion x: 5.0 ?
   -->Yes, But for the flight control, OSDK needs to continuously send control commands to the aircraft to ensure the aircraft fly. When using position control the drone, displacement processing may be required. You can refer to the demo ：demo_local_position_control.cpp
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/UsBXgGqCoHeziGyzSHhZqDbt0' target='zd_attach'>inline1106062784.png</a><br><a href='https://djisdksupport.zendesk.com/attachments/token/Tdn26YftaZXG6KBc4egLuor1Q' target='zd_attach'>inline-1430625720.png</a>

norihiro-ito, 2020-09-23:
DJI team, thank you very much for your continuous support. 
I gotcha from your answer.

1. The set_local_pos_ref() service set new origin from current_gps, And the coords dont change until reset by new call.

2. It should be so. But If a reached position reaves from the target position, I need to calculate the diff of both and resend new Joy topic.

If my understanding is wrong, please point out.
Thanks much.
nito

closed 2020-09-23 by norihiro-ito