dji-sdk/Onboard-SDK-ROS #397: [HELP] How to start "roslaunch dji_sdk sdk.launch" after boot in  Manifold2G

diogon01, 2020-09-23:
Dear friends,
I would like to know how to initialize the "**roslaunch dji_sdk sdk.launch**" after Manifold2G Boot. I tried to create the linux services with the sh file but apparently something went wrong.

I make the service file: **/etc/systemd/system/vant3d.service** and write: 
```
[Unit]
Description=start roscore
After=remote-fs.target
After=syslog.target
[Service]
ExecStart=/usr/local/bin/rosCoreService.sh
Restart=on-abort

[Install]
WantedBy=multi-user.target
```
This my bash file **/usr/local/bin/rosCoreService.sh**:

```
#!/bin/bash
source /home/bc/.bashrc
source /opt/ros/kinetic/setup.bash
source /home/bc/catkin_ws/devel/setup.bash
roslaunch multitower_rplidar rplidar.launch
```
Finally i'm add permission to execute my bash file and reload services:

```
sudo chmod +x  /usr/local/bin/rosCoreService.sh
sudo systemctl daemon-reload
sudo systemctl enable vant3d.service
sudo systemctl start vant3d.service

``` 
[image]

mhoejgaard, 2020-09-24:
I'd suggest using the robot_upstart package. 
http://wiki.ros.org/robot_upstart

diogon01, 2020-09-24:
> I'd suggest using the robot_upstart package.
> http://wiki.ros.org/robot_upstart
Dear, I read the documentation before your answer and it was not clear how to use the robot_upstart package,
Could you explain how to use it?

mhoejgaard, 2020-09-25:
I haven't used the package for a while but I believe this should work:
`$ rosrun robot_upstart install PATH_TO_LAUNCHFILE/file.launch`

It's borrowed from the documentation: http://docs.ros.org/jade/api/robot_upstart/html/

deadman2000, 2020-10-17:
I recommend this solution: https://blog.roverrobotics.com/how-to-run-ros-on-startup-bootup/