dji-sdk/Onboard-SDK-ROS #398: No speed control without downward vision positioning

carriojazz, 2020-09-24:
Hi, I am using OSDK 4.0.1 with the obsoleted interface to send speed commands to a M210v2 RTK using the topic:  
/dji_osdk_ros/flight_control_setpoint_generic

However, I have noticed that commands are ignored if downward vision positioning is disabled. Is this an expected behavior?

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39503'>Zendesk ticket #39503</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Yes，At present when downward vision positioning is disabled, the flight control command can't control the drone to descend. And this issue has been submitted to relevant engineer for further comfirmation, you can continue to follow the OSDK related function updates.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

mhoejgaard, 2020-09-25:
@dji-dev Is this only the case for descend commands?

carriojazz, 2020-09-26:
In my case it happens with all commands sent through /dji_osdk_ros/flight_control_setpoint_generic

dji-dev, 2020-09-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39503'>Zendesk ticket #39503</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The ROS can invoke the OSDK API to landing(comfirm landing),
You can refer to the code：vehicle_wrapper.cpp，goHomeAndConfirmLanding
 [image]​

Note: If you verify this function in real flight, it is recommended to ensure the safety of real flight

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/DUM3bDtXXLItC1g45npRL1ua2' target='zd_attach'>inline-1869952777.png</a>

mhoejgaard, 2020-09-27:
@dji-dev The question was whether it is expected behavior to block all commands sent through /dji_osdk_ros/flight_control_setpoint_generic when downward vision positioning is disabled?

dji-dev, 2020-09-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39503'>Zendesk ticket #39503</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

As mentioned above, whether this behavior is normal or not has been submitted for further confirmation, and the above-mentioned solution can be used to evaluate whether it can meet the demand.

Sorry for the inconvenience to you.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-09-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39503'>Zendesk ticket #39503</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

About the issue： The question was whether it is expected behavior to block all commands sent through  /dji_osdk_ros/flight_control_setpoint_generic when downward vision positioning is disabled?
-->The topic: /dji_osdk_ros/flight_control_setpoint_generic from the demo uses VERTICAL_VELOCITY to control the aircraft. downward vision positioning is disabled, the parameter: VERTICAL_THRUST is required. If you are using the origin demo, you may need to modify the relevant code.

In addition:
1. Make a precise landing: you can turn off the down view, but do not call the landing function when landing, you can use flightctrl to control the throttle to achieve, the interface VERTICAL_THRUST
2. Ordinary landing: It is not recommended to close the downward vision. If you encounter a hovering 0.7 situation, please use the remote control to confirm, or if you don't want to use the remote control, you can call the ForceLanding and ConfirmLanding API of OSDK(as mentioned above) to achieve landing.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team