dji-sdk/Onboard-SDK-ROS #399: failed when run sample node with version 4.0.1 in M300

dangviethieu, 2020-09-24:
hello there. I updated SDK to version 4.0.1, but when tested with M300, I got some issue
telemetry node: cannot get gps_health, gps_position
[image]
waypointV2_node: init mission failed
[image]
please help me check it 🙇

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39509'>Zendesk ticket #39509</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

1、Please confirm whether the GPS satellites can be searched normally. or it can be tested with simulator started.

2、Is the USB cable connected? waypoint v2 works with USB communication.
      or you can provide complete terminal logs to comfirm if the ACM device initial failed.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-09-25:
@dji-dev I tested with simulator, this is log of dji_osdk_ros node
[image]
As I see, current GPS health is 5, but telemetry node cannot get this value.

lorenzave, 2020-09-25:
For some reason the new wrapper (dji_vehicle) advertises the ROS topics but doesn't actually publish the messages, I recommend using the old interface if you don't want to code the publishing yourself(shouldn't be really complicated) but I hope DJI releases an actually working new interface. Its also missing velocity setpoint ROS subscribers, so you can only control the drone using their waypoint missions.

Best regards.

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39509'>Zendesk ticket #39509</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I test it with M300 (firmware: v01.00.211),  OSDK 4.0.1 , ROS 4.0.1 and simulator.
The telemetry node can get the GPS value, as follow result
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/RvkzzsMHa0jAyI7j6JmBDQc5O' target='zd_attach'>inline-2122477051.png</a>

lorenzave, 2020-09-25:
@dji-dev can you do a rostopic echo/hz to /dji/gps_position? we get no values whatsoever there. We do get them using the old interface.

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39509'>Zendesk ticket #39509</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

rostopic echo dji_osdk_ros/gps_position
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/qxYLDy1bRNB2dXNDhVzIqdfRK' target='zd_attach'>inline682173566.png</a>

dangviethieu, 2020-09-29:
@dji-dev I tested it in VMWare, maybe this make the issue. Have you try it?

dji-dev, 2020-09-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39509'>Zendesk ticket #39509</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Yes, I am verifying in VMware ubuntu1604, this seems to have nothing to do with whether or not to use VMware, I suggest you confirm to test and verify according to the version information I provided above.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-09-29:
@dji-dev I confirmed that I used M300 (firmware: v01.00.211), OSDK 4.0.1 , ROS 4.0.1
Maybe the problems is baud_rate, only 230400 working for me, other values make the issue: drone version not obtained! (I also changed value in DJI assistant)
But the recommend at this page: https://developer.dji.com/onboard-sdk/documentation/quickstart/run-the-sample.html
baud_rate should is 921600. What value did you use?

dji-dev, 2020-09-29:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39509'>Zendesk ticket #39509</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The baud rate may affect telemetry. ROS recommends using 921600 and above. I use 921600. To modify the baud rate, you need to modify the drone (DJI Assistant2) and the ROS configuration file (rosed dji_osdk_ros dji_vehicle_node.launch) at the same time. Modifying the baud rate of the drone requires restarting the drone.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-09-30:
@dji-dev I changed baud_rate to 921600 and that 's working  
I’m much obliged to you for your patient explanation 🙇

closed 2020-09-30 by dangviethieu