dji-sdk/Onboard-SDK-ROS #400: Failsafe error when OSDK disconected in 3.8.1

Xachap, 2020-09-24:
My development team and I have found a non-desirable behaviour of the ROS-OSDK about Failsafe. Actually, we are developing a system (using Manifold 2) for avoiding obstacles with the A3 PRO controller, and in DJI Assistant 2 the option “Failsafe SDK” is activated and in hovering mode. Despite of this, if there is a communication fail between Manifold and A3, it is not detected by the flight controller, and it continues with the mission instead of starting hovering. This causes a serious failure in our use case because it is not possible to continue without the avoid obstacle functionality given by our sensors (due to collision risk), and it is impossible for the pilot in order to identificate it. We are using ROS-OSDK 3.8.1 version, and we don't have checked it in the current one.

### How to reproduce this error: 
Executing, for example, the _demo_mission_ from _dji_sdk_demo_ with the DJI Assistant 2 simulator. In the middle of the execution, unplug the communication cable between the Manifold and the A3, or disconnect the power adaptor of the Manifold. The simulated drone continue the mission instead of starting hovering.

### Possible solution:
Add communication control from the A3 to the OSDK (check at 1Hz or 2Hz must be enought)

dji-dev, 2020-09-25:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39522'>Zendesk ticket #39522</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

This is because the waypoint mission is not controlled by OSDK. After OSDK uploads all mission points, the subsequent control actions are completed by the flight controller. Therefore, the OSDK needs to actively stop the task during the execution of the task, or use the remote control to obtain control and then control it.

If you do not use the waypoint mission but use the flightctl command, if the control command is not sent continuously, the aircraft will hover.
flightctl  API: flightCtrl (dji_control.hpp)

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

Xachap, 2021-02-07:
And in new models as M210 or M300, is this feature active?

I think that could be interesting to check if there is a problem in communications or in the power supply of the Manifold 2. Thank you again.