dji-sdk/Onboard-SDK-ROS #401: OSDK ROS v3.8: M600Pro => Topic 0x26 value memory not initialized

tkazik, 2020-09-26:
Dear SDK Support Team

Our setup:

- OSDK ROS: [v3.8](https://github.com/dji-sdk/Onboard-SDK-ROS/tree/3.8) (the latest one containing this [PR](https://github.com/dji-sdk/Onboard-SDK-ROS/commit/705d9c364858497e4ddecae5b7d9d168fa8c1b5b); not the one tagged with 3.8.1)
- M600 Pro (PM820V3PRO), Firmware (3.2.41.14)
- Tested in simulator of Assistant 2 for Matrice (V2.0.12)

Building runs without errors, but when running `roslaunch dji_sdk sdk.launch`, I get the following output:

```
STATUS/1 @ verify, L247: Verify subscription successful.
STATUS/1 @ startPackage, L353: Start package 2 result: 0.
STATUS/1 @ startPackage, L355: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L353: Start package 1 result: 0.
STATUS/1 @ startPackage, L355: Package 1 info: freq=50, nTopics=10.

STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=5, nTopics=1.

STATUS/1 @ removePackage, L471: Remove package 0 successful.
STATUS/1 @ startPackage, L353: Start package 0 result: 0.
STATUS/1 @ startPackage, L355: Package 0 info: freq=5, nTopics=11.

STATUS/1 @ startPackage, L353: Start package 3 result: 0.
STATUS/1 @ startPackage, L355: Package 3 info: freq=400, nTopics=1.

ERRORLOG/1 @ getValue, L315: Topic 0x26 value memory not initialized, return default[FATAL] [1601133098.020461943]: ASSERTION FAILED
	file = /opt/ros/noetic/include/ros/publisher.h
	line = 104
	cond = false
	message = 
[FATAL] [1601133098.020545752]: Call to publish() on an invalid Publisher
[FATAL] [1601133098.020594964]: 

[dji_sdk-2] process has died [pid 115298, exit code -5,

```
This issue is strongly related to https://github.com/dji-sdk/Onboard-SDK-ROS/issues/307 where I added a [comment](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/307#issuecomment-699503183) at the end, which contains plenty more information. Please fix...thank you!

dji-dev, 2020-09-27:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39575'>Zendesk ticket #39575</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The reason for this problem is that the M600 pro model and firmware do not support TOPIC: TOPIC_RTK_CONNECT_STATUS.
Sorry for the trouble caused by this problem.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

tkazik, 2020-09-29:
Thank you for you quick reply.
I would be nice, if you could create a branch for the M600, which works out of the box and one doesn't need to be patched.
Also, a clearer documentation about the actually supported topics would be nice (see e.g. https://github.com/dji-sdk/Onboard-SDK/issues/467 or https://github.com/dji-sdk/Onboard-SDK/issues/392).

dji-dev, 2020-09-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39575'>Zendesk ticket #39575</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Thank you very much for your suggestions, and we will continue to evaluate and improve the relevant information.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

fernandoespinosa1, 2020-12-03:
@tkazik I am using the 3.8.1 version(tag), and have pretty much the same issue, but with this output>

```
STATUS/1 @ verify, L244: Verify subscription successful.
STATUS/1 @ startPackage, L350: Start package 2 result: 0.
STATUS/1 @ startPackage, L352: Package 2 info: freq=100, nTopics=3.

STATUS/1 @ startPackage, L350: Start package 1 result: 0.
STATUS/1 @ startPackage, L352: Package 1 info: freq=50, nTopics=10.

STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=5, nTopics=1.

STATUS/1 @ removePackage, L468: Remove package 0 successful.
STATUS/1 @ startPackage, L350: Start package 0 result: 0.
STATUS/1 @ startPackage, L352: Package 0 info: freq=5, nTopics=11.

STATUS/1 @ startPackage, L350: Start package 3 result: 0.
STATUS/1 @ startPackage, L352: Package 3 info: freq=400, nTopics=1.

[FATAL] [1607019698.928078202]: ASSERTION FAILED
        file = /opt/ros/melodic/include/ros/publisher.h
        line = 103
        cond = false
        message =
[FATAL] [1607019698.929051952]: Call to publish() on an invalid Publisher
[FATAL] [1607019698.929940650]:

[dji_sdk-2] process has died [pid 144, exit code -6, cmd /opt/ros/melodic/devel/lib/dji_sdk/dji_sdk_node __name:=dji_sdk __log:=/root/.ros/log/5bf2b8c4-3594-11eb-9f68-b827ebb57be3/dji_sdk-2.log].
log file: /root/.ros/log/5bf2b8c4-3594-11eb-9f68-b827ebb57be3/dji_sdk-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
root@pi-desktop:/opt/ros/melodic#
```

As you can see, the only difference is that It is not displaying the Topic 0x26 value memory not initialized error before the first FATAL, and the number of line (103 in my case). I assume it is the same case than yours (the stuff related to TOPIC_RTK_CONNECT_STATUS). Do you know how can I solve it? I reviewed your well explained [comment here](https://github.com/dji-sdk/Onboard-SDK-ROS/issues/307#issuecomment-699503183)  and I am wondering how actually should I make this to work? It is something I can patch in the 3.8.1 branch to make this to work? I apprecciate any guidance on this issue.

tkazik, 2020-12-06:
Hi @fernandoespinosa1 

Unfortunately, the docs and support for the M600(pro) and ROS-SDK3.x are quite messy and misleading. I wasn't able to get the topic `TOPIC_RTK_CONNECT_STATUS` up and running. I tried several things but the code only worked without that topic. The dev team also sates above that the firmware does not support that topic (see: https://github.com/dji-sdk/Onboard-SDK-ROS/issues/401#issuecomment-699593205).
So my conclusion is that following: If DJI does not release a newer firmware for the M600, the topic `TOPIC_RTK_CONNECT_STATUS` will not be accessible...and I would be very surprised if they did release another firmware upgrade, given the 'age' of the M600(pro). Sorry for not having better news...if you figure out something please let the community know. Good luck!

LiamHorton, 2022-10-06:
Has there been any update on this issue? Currently facing the same problem. Thanks

snakehaihai, 2023-10-23:
same guess ill try to downgrade to much earlier version

snakehaihai, 2023-10-23:
solved it. go to  src/module/dji_sdk_node.cpp

search term RTK
comment out all sections related to RTK. then can compile and run