dji-sdk/Onboard-SDK-ROS #402: Waypoint Action (Gimbal, AircraftControl) failed

dangviethieu, 2020-10-02:
I run sample waypointV2_node, changed **_waypointV2ACtionActuatorType_** to **_DJIWaypointV2ActionActuatorTypeAircraftControl_**, but got the error like:
[image]
I try with **_DJIWaypointV2ActionActuatorTypeGimbal_** also got the same error
Do you have any idea what may be happening? Is this the bug? Or we didn't implement this function yet.
My code:
> bool generateWaypointV2Actions(ros::NodeHandle &nh, uint16_t actionNum)
{
        waypointV2_generate_actions_client = nh.serviceClient<dji_osdk_ros::GenerateWaypointV2Action>("dji_osdk_ros/waypointV2_generateActions");
        dji_osdk_ros::WaypointV2Action actionVector;
        for (uint16_t i = 0; i < actionNum; i++)
        {
      actionVector.actionId  = i;
      actionVector.waypointV2ActionTriggerType  = dji_osdk_ros::WaypointV2Action::DJIWaypointV2ActionTriggerTypeSampleReachPoint;
      actionVector.waypointV2SampleReachPointTrigger.waypointIndex = i;
      actionVector.waypointV2SampleReachPointTrigger.terminateNum = 0;
      actionVector.waypointV2ACtionActuatorType = dji_osdk_ros::WaypointV2Action::DJIWaypointV2ActionActuatorTypeAircraftControl;
      actionVector.waypointV2AircraftControlActuator.actuatorIndex = 0;
      actionVector.waypointV2AircraftControlActuator.DJIWaypointV2ActionActuatorAircraftControlOperationType = dji_osdk_ros::WaypointV2AircraftControlActuator::DJIWaypointV2ActionActuatorAircraftControlOperationTypeRotateYaw;
      actionVector.waypointV2AircraftControlActuator.waypointV2AircraftControlActuatorRotateHeading.isRelative = true;
      actionVector.waypointV2AircraftControlActuator.waypointV2AircraftControlActuatorRotateHeading.yaw = 60;
      generateWaypointV2Action_.request.actions.push_back(actionVector);
    }
    waypointV2_generate_actions_client.call(generateWaypointV2Action_);
    return generateWaypointV2Action_.response.result;
}

dji-dev, 2020-10-03:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39708'>Zendesk ticket #39708</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

We have found the problem in waypoint v2 and submitted it to the relevant engineer for confirmation and repair. If there is any progress, we will communicate with you in time.
Sorry for the inconvenience caused to your related development work. And suggest you continue to pay attention to the updates of DJI SDK related functions.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dji-dev, 2020-10-14:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39708'>Zendesk ticket #39708</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

I'm sorry that the problem has kept you waiting for a long time. The problem has been fixed and updated to the Git Hub master branch. It is recommended that you can update the OSDK from the master branch to verify this problem.

https://github.com/dji-sdk/Onboard-SDK
 [image]​

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-10-21:
@dji-dev Thank for reply, I confirm that The actions has been work.
But with **_WaypointV2AircraftControlActuatorFlying_.msg**, 
`uint8  isStartFlying  # Determines the aircraft start flying or stop flying. 
 ``TRUE`` for the aircraft to start flying.
`
follow my understanding, this is actions for drone hovering at waypoint. like ver3.8
[image]
So, whether we miss the param: **hover_time**?

dji-dev, 2020-10-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/39708'>Zendesk ticket #39708</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Sorry, the current parameter: isstartflying cannot set the hover time, it only provides settings to continue flying or stop flying. If you need to set hover, you can consider using pauseWaypointMission and resumeWaypointMission.

If the above cannot meet your needs, It is recommended that you click on the links on the right side and fill in the relevant demand information,  https://formcrafts.com/a/dji-developer-feedback-en
In order to better match your requirements, try to include the following information when filling in：
1. Desired functions and requirements
2. Company/organization profile
3. Project descriptions that use this feature requirement.
4. Personal background or brief introduction.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-10-29 by dangviethieu