dji-sdk/Onboard-SDK-ROS #405: Mission waypoint action doesn't work (M210 V2 RTK)

manfer33, 2020-10-15:
I am using v4.0.1 of OSDK with M210 V2 RTK. I am trying to execute a mission waypoint (v1) where I would like to do an action in the first and last waypoint of the list. To do it I am testing just applying the wait option of the command list. The first waypoint is setted like this but no action seems to be applied while the mission is running: 

```
...  
  waypointTask.mission_waypoint[0].has_action = 1;
  waypointTask.mission_waypoint[0].waypoint_action.action_repeat = 17; // 00010001 - 1 action, 1 time
  waypointTask.mission_waypoint[0].waypoint_action.command_list[0] = 0; // WP_ACTION_STAY = 0
  waypointTask.mission_waypoint[0].waypoint_action.command_parameter[0] = 5000; // ms
....
```

Have anyone any idea about the cause of this? Did I forget any parameter?

dji-dev, 2020-10-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40063'>Zendesk ticket #40063</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

According to the information you provided, set the action to WP_ACTION_STAY,
this will not execute the action, and you can refer to the comment of the relevant action in the code.

dji_mission_type.hpp
typedef enum WaypointActionTypeFormat
{
  WP_ACTION_STAY                = 0,  /*!< no action.uint of action parameter:ms*/
  WP_ACTION_SIMPLE_SHOT          = 1,  /*!< take picture action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_VIDEO_START          = 2,  /*!< start take video action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_VIDEO_STOP          = 3,  /*!< stop video action.action parameters Action parameter have no effect.limit time:6s*/
  WP_ACTION_CRAFT_YAW            = 4,  /*!< craft control yaw action.uint of action parameter:degree. range:-180 ~ 180*/
  WP_ACTION_GIMBAL_PITCH        = 5,  /*!< gimbal control pitch action.uint of action parameter:degree. range:-90 ~ 0*/
} WaypointActionTypeFormat;

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

manfer33, 2020-10-16:
Hello, thank you for your response. If you see in my code the following line:

`  waypointTask.mission_waypoint[0].waypoint_action.command_list[0] = 0; // WP_ACTION_STAY = 0
`
I am already using the WP_ACTION_STAY in command_list with its value 0 and as I said no action seems to be applied while the mission is running.

Regards

dji-dev, 2020-10-16:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40063'>Zendesk ticket #40063</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Because you set waypointTask.mission_waypoint[0].waypoint_action.command_list[0] = 0; // WP_ACTION_STAY = 0,
and WP_ACTION_STAY means no action, as follows:
  WP_ACTION_STAY = 0, /*!< no action.uint of action parameter:ms*/

In addition, I’m sorry that I didn’t fully understand the "no action seems to be applied while the mission is running" you mentioned. Can you elaborate, for example, what action do you need to perform?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

manfer33, 2020-10-16:
Hello again,

I know what WP_ACTION_STAY means. I am trying just to set a wait task along some seconds. 

As you can see in the following line of code (same of the first message), I am setting that time to wait, but when the uav reach the waypoint doesn't wait the time setted.

`  waypointTask.mission_waypoint[0].waypoint_action.command_parameter[0] = 5000; // ms
`
Regards

dji-dev, 2020-10-22:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40063'>Zendesk ticket #40063</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Okay, I understand your question. According to the code parameter settings you provided, there are some problems:
Regarding the parameter: waypoint_action.action_repeat, the 17 you set has overflowed.

dji_mission_type.hpp
 [image]​

It is recommended that you can test whether it is normal according to the following settings

  wp->damping        = 0;
  wp->yaw            = 0;
  wp->gimbalPitch    = 0;
  wp->turnMode        = 0;
  wp->hasAction      = 1;
  wp->actionTimeLimit = 100;
  wp->actionNumber    = 6;
  wp->actionRepeat    = 1;
  for (int i = 0; i < 16; ++i)
  {
    wp->commandList[i]      = WP_ACTION_STAY;
    wp->commandParameter[i] = 2000;
  }

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/drcC4v511xdwwzLyHSexJhG79' target='zd_attach'>inline-652177043.png</a>