dji-sdk/Onboard-SDK-ROS #410: Yaw Actions (DJIWaypointV2ActionActuatorAircraftControlOperationTypeRotateYaw)

dangviethieu, 2020-10-30:
Hi all. I created plan for wp v2 with yaw actions at each wp. I want heading of drone is always point to a diff constant value at each wp. But reality, drone only rotate to heading when reach at wp, then it auto rotate to default heading. This is my code.
Do you have any idea what may be happening? Anyone can help me? Thank you in advance :) 
`dji_osdk_ros::WaypointV2Action actionVector;
actionVector.actionId = ros::Time::now().toSec();
actionVector.waypointV2ActionTriggerType  = dji_osdk_ros::WaypointV2Action::DJIWaypointV2ActionTriggerTypeSampleReachPoint;
actionVector.waypointV2SampleReachPointTrigger.waypointIndex = index;
actionVector.waypointV2SampleReachPointTrigger.terminateNum = 0;
actionVector.waypointV2ACtionActuatorType = dji_osdk_ros::WaypointV2Action::DJIWaypointV2ActionActuatorTypeAircraftControl;
actionVector.waypointV2AircraftControlActuator.actuatorIndex = 0;
actionVector.waypointV2AircraftControlActuator.DJIWaypointV2ActionActuatorAircraftControlOperationType = dji_osdk_ros::WaypointV2AircraftControlActuator::DJIWaypointV2ActionActuatorAircraftControlOperationTypeRotateYaw;
actionVector.waypointV2AircraftControlActuator.waypointV2AircraftControlActuatorRotateHeading.isRelative = false;
actionVector.waypointV2AircraftControlActuator.waypointV2AircraftControlActuatorRotateHeading.yaw = index * 20;`

dji-dev, 2020-10-30:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40588'>Zendesk ticket #40588</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Have you tried setting waypointV2.headingMode? Does this meet your application scenario?

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

dangviethieu, 2020-11-01:
@dji-dev Thank for suggesting, I try with all **headingMode** but it doesn't respond my purpose.
In previous versions, we had WP_ACTION_CRAFT_YAW for adjust aircraft heading when reach at WP.
[image]

dangviethieu, 2020-11-09:
@dji-dev thank you, I changed headingMode and fixed the issue

closed 2020-11-09 by dangviethieu