dji-sdk/Onboard-SDK-ROS #411: Error when attempting to control gimbal

lorenzmhe, 2020-11-03:
In OSDK 4.0.1 and OSDK-ROS 4.0.1 I sometimes get the following error when attempting to control the gimbal with the /task_gimbal_control rosservice. The error is printed and the gimbal does not move no matter what command is sent. Only a reboot fixes the issue.

```
$ roslaunch dji_osdk_ros dji_vehicle_node.launch 
... logging to /home/mathias/.ros/log/fbb2d3e0-1dc9-11eb-ae19-0c9d92105bd2/roslaunch-lorenz-mhe-16813.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lorenz-mhe:45393/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.16
 * /vehicle_node/acm_name: /dev/ttyACM0
 * /vehicle_node/align_time: False
 * /vehicle_node/app_id: 1055958
 * /vehicle_node/app_version: 1
 * /vehicle_node/baud_rate: 921600
 * /vehicle_node/dxc: False
 * /vehicle_node/enc_key: XXX
 * /vehicle_node/serial_name: /dev/ttyUSB0
 * /vehicle_node/use_broadcast: False

NODES
  /
    vehicle_node (dji_osdk_ros/dji_vehicle_node)

auto-starting new master
process[master]: started with pid [16823]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to fbb2d3e0-1dc9-11eb-ae19-0c9d92105bd2
process[rosout-1]: started with pid [16837]
started core service [/rosout]
process[vehicle_node-2]: started with pid [16854]
EnableAd: 1
[2381090.573]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!
[2381090.597]STATUS/1 @ parseDroneVersionInfo, L1122: Device Serial No. = 1DADG4J00100KQ
[2381090.597]STATUS/1 @ parseDroneVersionInfo, L1124: Firmware = 3.4.3.44
[2381090.597]STATUS/1 @ functionalSetUp, L279: Shake hand with drone successfully by getting drone version.
[2381090.597]STATUS/1 @ legacyX5SEnableTask, L56: Legacy X5S Enable task created.
[2381091.597]STATUS/1 @ sendHeartbeatToFCTask, L1546: OSDK send heart beat to fc task created.
[2381091.797]STATUS/1 @ FileMgrImpl, L253: register download file callback handler successfully.
[2381091.798]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2381091.798]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2381091.798]STATUS/1 @ PSDKModule, L98: MOP only support M300, so mop client will not be initialized here.
[2381091.798]STATUS/1 @ RegisterOSDInfoCallback, L471: register result of geting　FC ground station status pushing : 1

[2381091.798]STATUS/1 @ initDJIHms, L900: DJI HMS is not supported on this platform!

[2381091.798]STATUS/1 @ init, L247: USB is not plugged or initialized successfully. Advacned-Sensing will not run.
[2381091.798]STATUS/1 @ activate, L1329: version 0x304032C

[2381091.804]STATUS/1 @ activate, L1353: Activation successful

[ INFO] [1604403894.245642386]: VehicleNode Start
[ INFO] [1604403894.245827342]: Topic startup!
[ INFO] [1604403894.276977079]: Services startup!
[ INFO] [1604403894.288572951]: Use data subscription to get telemetry data!
[ INFO] [1604403894.288590345]: align_time_with_FC set to false. We will use ros time to time stamp messages!
[2381092.905]STATUS/1 @ verify, L244: Verify subscription successful.
[2381093.001]STATUS/1 @ startPackage, L345: Start package 3 result: 0.
[2381093.001]STATUS/1 @ startPackage, L347: Package 3 info: freq=100, nTopics=3.

[2381093.039]STATUS/1 @ startPackage, L345: Start package 2 result: 0.
[2381093.039]STATUS/1 @ startPackage, L347: Package 2 info: freq=50, nTopics=14.

[2381093.061]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[2381093.061]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=1.

[2381093.110]STATUS/1 @ removePackage, L457: Remove package 1 successful.
[2381093.141]STATUS/1 @ startPackage, L345: Start package 1 result: 0.
[2381093.141]STATUS/1 @ startPackage, L347: Package 1 info: freq=5, nTopics=12.

[2381093.165]STATUS/1 @ startPackage, L345: Start package 4 result: 0.
[2381093.165]STATUS/1 @ startPackage, L347: Package 4 info: freq=400, nTopics=1.

[ INFO] [1604403924.377024398]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 61.20)
[ INFO] [1604403924.377067700]: Call gimbal Ctrl.
[2381122.969]STATUS/1 @ rotateGimbal, L1816: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,90.00) mode:1 time:1.00s
[2381122.969]ERRORLOG/1 @ rotateGimbal, L1821: Rotation gimbal fail. Error code : 0xFFFFFFFF
[2381122.969]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[2381122.969]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[2381122.969]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[ INFO] [1604403926.377426964]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 61.20)
[ INFO] [1604403972.108379437]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604403972.108411645]: Call gimbal Ctrl.
[2381170.700]STATUS/1 @ rotateGimbal, L1816: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:1 time:1.00s
[2381170.700]ERRORLOG/1 @ rotateGimbal, L1821: Rotation gimbal fail. Error code : 0xFFFFFFFF
[2381170.700]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[2381170.700]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[2381170.700]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[ INFO] [1604403974.108625103]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604403988.285215764]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604403988.285253449]: Call gimbal Ctrl.
[2381186.877]STATUS/1 @ rotateGimbal, L1816: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:0 time:1.00s
[2381186.877]ERRORLOG/1 @ rotateGimbal, L1821: Rotation gimbal fail. Error code : 0xFFFFFFFF
[2381186.877]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[2381186.877]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[2381186.877]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[ INFO] [1604403990.285369215]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604404249.884044710]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604404249.884100864]: Call gimbal Ctrl.
[2381448.476]STATUS/1 @ rotateGimbal, L1816: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:0 time:1.00s
[2381448.476]ERRORLOG/1 @ rotateGimbal, L1821: Rotation gimbal fail. Error code : 0xFFFFFFFF
[2381448.476]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[2381448.476]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[2381448.476]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[ INFO] [1604404251.884337227]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604404257.339990157]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 87.10)
[ INFO] [1604404257.340022567]: Call gimbal Ctrl.
[2381455.932]STATUS/1 @ rotateGimbal, L1816: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:0 time:1.00s
[2381455.932]ERRORLOG/1 @ rotateGimbal, L1821: Rotation gimbal fail. Error code : 0xFFFFFFFF
[2381455.932]ERRORLOG/1 @ printErrorCodeMsg, L931: >>>>Error module   : System
[2381455.932]ERRORLOG/1 @ printErrorCodeMsg, L932: >>>>Error message  : Undefined error
[2381455.932]ERRORLOG/1 @ printErrorCodeMsg, L933: >>>>Error solution : Unknown error code : 0X%lX, please contact <dev@dji.com> for help.
[ INFO] [1604404259.340254032]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 87.10)

```

ShreyasManjunath, 2020-11-03:
Hello Lorenzmhe,

May I know how the Gimbal service is called? Through rosservice call or from ROS Node with ServiceClient?

[(https://github.com/dji-sdk/Onboard-SDK-ROS/blob/master/src/dji_osdk_ros/samples/gimbal_camera_control_node.cpp)]

You can refer to this.

Regards,

Shreyas

lorenzave, 2020-11-03:
HI @ShreyasManjunath,
we have seen it happen with both approaches. :-1: 

Once it gets there we cannot recover unless we reboot the drone.

ShreyasManjunath, 2020-11-03:
Hi,

I was curious about seeing your observation.

Few hours ago, Found these results.

[ INFO] [1604415458.048524288]: Current gimbal 0, angle (p,r,y) = (0.00, 0.10, 89.90)
[ INFO] [1604415458.048687936]: Call gimbal Ctrl.
[2392656.640]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,0.00) mode:1 time:0.50s
[ INFO] [1604415460.058066592]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415460.064588992]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415460.064738848]: Call gimbal Ctrl.
[2392658.656]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,0.00) mode:1 time:0.50s
[ INFO] [1604415462.072898144]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415462.076547840]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415462.076652384]: Call gimbal Ctrl.
[2392660.668]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,0.00) mode:1 time:0.50s
[ INFO] [1604415464.092522720]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415471.861325568]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 0.00)
[ INFO] [1604415471.861475808]: Call gimbal Ctrl.
[2392670.453]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,90.00) mode:1 time:0.50s
[ INFO] [1604415473.877222496]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.00, 89.90)
[ INFO] [1604415473.883354656]: Current gimbal 0, angle (p,r,y) = (0.00, 0.00, 89.90)
[ INFO] [1604415473.883506848]: Call gimbal Ctrl.
[2392672.475]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,90.00) mode:1 time:0.50s
[ INFO] [1604415475.892462528]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.10, 89.90)
[ INFO] [1604415484.143659552]: Current gimbal 0, angle (p,r,y) = (0.00, 0.10, 89.90)
[ INFO] [1604415484.143812384]: Call gimbal Ctrl.
[2392682.735]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:1 time:0.50s
[ INFO] [1604415486.168315680]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.10, 70.00)
[ INFO] [1604415486.173367904]: Current gimbal 0, angle (p,r,y) = (0.00, 0.10, 70.00)
[ INFO] [1604415486.173515744]: Call gimbal Ctrl.
[2392684.765]STATUS/1 @ rotateGimbal, L1844: Start to rotate the gimbal 0, (p,r,y) = (0.00,0.00,70.00) mode:1 time:0.50s
[ INFO] [1604415488.183844960]: Current gimbal 0 , angle (p,r,y) = (0.00, 0.10, 70.00)

dji-dev, 2020-11-04:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40706'>Zendesk ticket #40706</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

The log shows that the ACM device connections abnormal.
 [image]​

Please check the dual-A USB cables connects the drone and Onboard computer , then ping 192.168.42.2
If the network is unavailable，please check the ACM device on Onboard Computer and config the ip : 192.168.42.3 (for M210 V2)

For OSDK 4.x version ,The camera manager, gimbal manager and advance sensing features need the USB communication.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team<br><a href='https://djisdksupport.zendesk.com/attachments/token/nDPwBaoA2env164Q4G1OpEvE2' target='zd_attach'>inline831524784.png</a>

lorenzmhe, 2020-11-04:
USB was connected and network was fine.

dji-dev, 2020-11-05:
Agent comment from kyle.cai in <a href='https://djisdksupport.zendesk.com/agent/tickets/40706'>Zendesk ticket #40706</a>:<br><br>Dear developer ,

Thank you for contacting DJI.

Will this error occur when “USB was connected and network was fine.” ？
[2381090.573]ERRORLOG/1 @ initVehicle, L162: Failed to initialize ACM Linker channel!

If so，It may be necessary to check the access permission of the ACM device, try with root user.

Thank you for your understanding and support, hope you have a nice day.

Best Regards,
DJI SDK Support Team

closed 2020-12-04 by lorenzmhe